wb2osz/direwolf #1: IGate Stats to Telemetry Data

kernworks, 2015-08-11:
Hello.

First off, great application! I am using it as a RX IGate on a Raspberry Pi.

My question, I see that the IGate module has stat counters (line 282 to 315 in igate.c). Is there a way to send those stats along with the PBEACON of the IGate?

I see the TODO on line 284 already so I am guessing this is more of a feature request.

Thank You.

wb2osz, 2015-08-19:
What would you like to see?

irl, 2016-04-29:
I'd love to see similar telemetry options to aprx.

http://ham.zmailer.org/oh2mqk/aprx/aprx-manual.pdf

Per-radio-interface statistics that can be sent on the selected interface/to APRS-IS.

+1 on this feature request.

bradanlane, 2016-06-21:
I have been playing with this idea. I coded an SBEACON option in the direwolf.conf which outputs packet stats to the internet. This way, the stats can be beaconed at its own interval.. 

An example beacon looks like:
`2016-06-21 13:32:39 EDT: N0ABC-1>APDW13,TCPIP*,qAC,THIRD:!3850.29ND07703.33W&Statistics: RX=12345 TX=0 UP=12123 DN=0`

This indicates N0ABC-1 heard 12345 packets on it's radio and after de-duplicaiton forwarded 12123 packets to the internet. Since this is an RX only iGate, 0 packets were transmitted and 0 packets were received from the internet.

Since it is its own beacon type, it can be set to its own interval such an hourly.

Is this along the lines of what people want?

kernworks, 2016-06-27:
That is basically what I was looking for. Awesome to see it happen.

My use for this is basically a packets per hour (or some set time period) set of counters that can be reported.

Personally, I would like counters to be set for "since last beaconing" but I could see how others may want since last start.

If you have some code to test, I'd be happy to do so. I'm running RX only as well (Raspberry Pi + SDR + Direwolf).

bradanlane, 2016-06-27:
Having run this code for a week and analyzing the usefulness of the four metrics, I've concluded a few things:

1. having the values be "since last beacon" is both more useful and significantly easier to maintain.
2. with "since last start", there must is extra code to track the datetime in order to reset at the start of the new month (to avoid eventual overflow)
3. the graphs I've found most useful depict the hourly traffic (I having the SBEACON set to hourly) so currently the spreadsheet is post processing the metrics to get "deltas".

I'll switch to "since last beacon", remote the datetime tracking and reset code. My current work is on the DEV branch. 

I'm loath to fork the entire project for these small improvements but unless I can work with WB2OSZ to get contributor status, I may not have a choice. If I do fork, I'll make it private so it doesn't wrench the Internet with people pulling from tangent repositories. I'll also delete the fork as soon as the pull-request is accepted.

dranch, 2016-06-27:
The popular UIView32 APRS program has a module that beacons the # of 
packets in the "last 10 minutes".  Now, I suppose you can tune the 
beacon interval to be every 10 minutes but some thought should go into 
this interval and how it's configured.  On classic KB2KB packet 
networks, beacons running every 10 minutes creates a lot of unnecessary 
traffic but being able to send out stats data like this would be cool if 
done properly.

--David

On 06/27/2016 02:30 PM, bradanlane wrote:
> Having run this code for a week and analyzing the usefulness of the
> four metrics, I've concluded a few things:
>
>  1. having the values be "since last beacon" is both more useful and [...]

bradanlane, 2016-06-27:
In direwolf, the beacon interval is user configurable. There are currently a number of different beacon-types and each may be configured with its own interval. My code adds a new beacon-type so it can be on whatever interval the operator wishes.

bradanlane, 2016-06-28:
It turns out there is no way to create a private fork of a public repository. Apologies to those who would like access to the SBEACON feature. I defer to @wb2osz on how this code can be integrated into direwolf.

wb2osz, 2016-07-03:
The latest version on the "dev" branch has a new IBEACON beacon type for sending IGate statistics.
The format is based on existing standards and previous precedent with a couple extra values added.   Details in the User Guide.

closed 2016-07-03 by wb2osz

bradanlane, 2016-07-03:
Thanks. I pull the latest and replace my implementation.