wb2osz/direwolf #2: Font color similar to background

DL7AD, 2015-08-28:
On Linux the background color on terminals is almost the same than the colors used for the fonts which makes it unreadable.
[image]

wb2osz, 2015-09-18:
A work-around is to use the "-t 0" command line option to disable text color.

See comments in textcolor.c for more discussion of this issue.

cniesen, 2017-11-19:
FYI, I encountered the same issue, wanting the multiple colors but a black background.
Since I was building from the source I swapped the color code of black and white in the `textcolor.c` file.  Looks nice now.

ko6kL, 2018-08-10:
-t o 
on the command line worked , thank you , the light green on white was killing me....
right click icon make property like this...
C:\DownLoads-\ham-programs\direwolf\direwolf-1.5-beta2-win\direwolf.exe -t 0

a larger box would be nice to keep things on one line ???

wb2osz, 2019-07-07:
Text color handling had a major overhaul in the "dev" branch in preparation for release 1.6.  it now works better with more terminal types.   Please give it a try and let me know how it works for you.

ko6kL, 2019-07-07:
have the colorsstart up text on the HF window ( top_) bottom in 2m window.

file attached.

Tim-VK3MIT, 2019-08-27:
> A work-around is to use the "-t 0" command line option to disable text color.
> 
> See comments in textcolor.c for more discussion of this issue.

How do i use the "-t 0" switch in the windows version of Direwolf (the text colours on the screen are a problem) 

I've tried modifying the direwolf.exe filename to include the switch as previously suggested, but windows then doesn't know how to handle the file. Where am I going wrong please?

Thank you - Tim.

wb2osz, 2020-03-21:
The "issues" section is for reporting software bugs and making new feature requests.
It is not a support forum for questions.  Please send questions to the discussion group.
https://groups.io/g/direwolf

closed 2020-03-21 by wb2osz