wb2osz/direwolf #4: GPIO pin naming

ohanssen, 2015-09-29:
When setting up GPIO for PTT Direwolf expects a number. This doesn't work on the Cubieboard where gpio pins are named differently. E.g. GPIO pin 61 is mapped to the device name gpio61_pi13 instead of just gpio61. To activate a pin, one still use just the  number. E.g. "echo 61 > /sys/class/gpio/export". 

Is it possible to support these boards by configuring using a name instead of just a number, or is there a workaround?

wb2osz, 2017-03-05:
This might have been fixed by 58c2707f7d65b5b074b5e67e00c06a50fb968349 on the 'dev' branch but I don't have a CubieBoard to verify.  See comments in ptt.c for details.
Would someone like to test this and report the results?
Just use the pin number like 61.  If the device name like gpio61 is not found, it will look for something like gpio61_*

wb2osz, 2017-04-01:
This should be fixed in the “dev” branch.

I don’t have a CubieBoard to test it.

Can you verify?

Thanks.

wb2osz, 2017-05-10:
This should be fixed in the most recent version.  (1.4)

I don’t have a CubieBoard to test it.

Can you verify?

Thanks.

closed 2017-05-10 by wb2osz