wb2osz/direwolf #6: audio_portaudio.c line 917, Abort trap : 6

M0CRT, 2015-10-29:
Hi

First time sending one of these!

Hope this helps us both.

Compiled OSX version using OSX makefile.  

Running OSX 10.11.

Had to ensure portaudio was installed as a UNIVERSAL install from mac ports in order to obtain the i386 libraries (stumped for a short time).

Apart from modifying the direwolf.conf for my callsign, I uncommented :

ADEVICE - plughw:1,0
ADEVICE UDP:7355 default

Upon execution of direwolf, see below:

Dire Wolf DEVELOPMENT version 1.3 F (Oct 29 2015)

Reading config file direwolf.conf
Audio input device for receive: UDP:7355  (channel 0)
Audio out device for transmit: default  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate.
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Use -p command line option to enable KISS pseudo terminal.
Assertion failed: (adev[a].inbuf_size_in_bytes >= 100 && adev[a].inbuf_size_in_bytes <= 32768), function audio_get, file audio_portaudio.c, line 917.
Abort trap: 6

Let me know if you need any logs etc.

Nigel

dranch, 2015-10-29:
Hello Nigel,

> Had to ensure portaudio was installed as a UNIVERSAL install from mac 
> ports in order to obtain the i386 libraries (stumped for a short time). 

Direwolf doesn't support PulseAudio natively.  It really uses ALSA 
directly but it can co-exist with Pulse as long as Pulse isn't using the 
device at the same time.

> Apart from modifying the direwolf.conf for my callsign, I uncommented :
>
> ADEVICE - plughw:1,0
> ADEVICE UDP:7355 default
>

Direwolf supports multiple concurrent soundcards / data streams but they 
must be unique.  You have specified "ADEVICE" twice which is conflicting 
but interestingly enough, it seems to take the UDP stream for input yet 
uses "Default" on output.  I'm not 100% sure what "Default" means here 
(it's not clearly documented) but I'm assuming it's using PulseAudio's 
"default".  This split device support is pretty slick but I'm not sure 
this is actually supported today (I'll check with John) .  If it's not 
supported, Direwolf should throw an error.

What you probably want is:

#Device #1 : taking it's input from STDIN using the "-"
ADEVICE - plughw:1,0

# Device #2 - taking an external input
ADEVICE1 UDP:7355 default

Yes?

> Upon execution of direwolf, see below:
>
> Dire Wolf DEVELOPMENT version 1.3 F (Oct 29 2015)
>
> Reading config file direwolf.conf
> Audio input device for receive: UDP:7355 (channel 0)
> Audio out device for [...]

M0CRT, 2015-10-29:
Hi Sir!

Thanks for the very rapid response!  Adjusted .conf as suggested.

#ADEVICE  plughw:1,0

# Starting with version 1.0, you can also use "-" or "stdin" to
# pipe stdout from some other application such as a software defined
# radio.  You can also specify "UDP:" and an optional port for input.
# Something different must be specified for output.

...

ADEVICE - plughw:1,0
#ADEVICE UDP:7355 default

#
# Number of audio channels for this souncard:  1 or 2.
#

ACHANNELS 1
#ACHANNELS 2

#############################################################
#                                                           #
#               SECOND AUDIO DEVICE PROPERTIES              #
#               (Channel 2 + 3 if in stereo)                #
#                                                           #
#############################################################

ADEVICE1 UDP:7355

...

If I comment out both, rightly, I get no audio devices available BUT I do get a list.

Dire Wolf DEVELOPMENT version 1.3 F (Oct 29 2015)

Reading config file direwolf.conf
Audio device for both receive and transmit:   (channel 0)
Number of devices = 7
--------------------------------------- device #0
[ Default Input ]
Name        = "Built-in Microph"
Host API    = Core Audio
Max inputs  = 2
Max outputs = 0
--------------------------------------- device #1
Name        = "Built-in Input"
Host API    = Core Audio
Max inputs  = 2
Max outputs = 0
--------------------------------------- device #2
[ Default Output ]
Name        = "Built-in Output"
Host API    = Core Audio
Max inputs  = 0
Max outputs = 2
--------------------------------------- device #3
Name        = "Adobe Encore DVD 6.0"
Host API    = Core Audio
Max inputs  = 0
Max outputs = 0
--------------------------------------- device #4
Name        = "After Effects 11.0.4"
Host API    = Core Audio
Max inputs  = 0
Max outputs = 0
--------------------------------------- device #5
Name        = "Prelude 1.0"
Host API    = Core Audio
Max inputs  = 0
Max outputs = 0
--------------------------------------- device #6
Name        = "Premiere Pro 6.0"
Host API    = Core Audio
Max inputs  = 0
Max outputs = 0
Input device name null
Pointless to continue without audio device.

Still same error...hmmm.

Any other thoughts?  Do I need to remove or change the audioport libraries?

Thanks

Nige

dranch, 2015-10-29:
Hello Nigel,

Starting at the beginning, what are you specifically trying to do? What 
are you trying to listen on and what are you trying to transmit on?

> Thanks for the very rapid response! Adjusted .conf as suggested. 
> ADEVICE - plughw:1,0

A critical piece of information missing here is how are you starting 
Direwolf.  Are you starting it similar to what's in the UserGuide like this?

    rtl_fm -f 144.39M -o 4 - | direwolf -n 1 -r 24000 -b 16

> ADEVICE1 UDP:7355 

Ok..

> ...
>
> If I comment out both, rightly, I get no audio devices available BUT I 
> do get a list.
>

Comment out both what?  ADEVICES?  CHANNELS?  Please be specific or 
better yet, please attach your complete config.

Try starting up Direwolf with *one* device at a time and get it work 
they way you expect them to work.  Once you have that right, then start 
trying to run multiple interfaces running in parallel.

> Reading config file direwolf.conf
>
> Audio device for both receive and transmit: (channel 0)
> Number of devices = 7
> --------------------------------------- device #0
> [ Default Input ]
> Name = "Built-in Microph"
> Host API = Core Audio
> Max inputs = 2
> Max outputs = 0
>

Please follow the UserGuide and use the "aplay" and "arecord" results to 
better pick the right ALSA devices.

--David

M0CRT, 2015-10-29:
Hi David.

Just a tinker at this end pending a more serious go at tomorrow...more likely on a Raspberry Pi 2.

Just wished you to be aware of the potential issue with the error I was receiving on run.

Attached my config.  

Was attempting to bring up the system first with just an execution of 'direwolf' to start with post compile.

Apologies if this 'error' I mentioned re the portaudio line 917 should be expected if I'm not executing in a pipe with the the rtl_ file.

Nige

wb2osz, 2015-10-30:
What we've got here is failure to communicate. 

Nigel is using the Mac OSX version. David is using the Linux version. 

The two versions have different ways of specifying the audio I/O devices. 

The User Guide has a short section, explaining the Mac OSX differences, which needs to be expanded and improved. 

The original error message, which started this discussion, needs to provide a better description of the problem.

wb2osz, 2015-12-26:
A fix for the "Assertion failed" in the Mac version is now in the "dev" branch.

closed 2015-12-26 by wb2osz