wb2osz/direwolf #8: debian packaging

elafargue, 2015-11-08:
It would be excellent to provide a debian directory for creating .deb packages.

ohanssen, 2015-12-23:
Direwolf is actually in Debian. However, the version that is available in Debian Stable is way too old. The version available in Debian Unstable is up-to-date.

irl, 2016-04-29:
If there is demand, I'm happy to provide backports of direwolf to Debian stable. I am also considering packaging beta versions of direwolf in the experimental suite of the archive (which could be installed using [apt pinning](http://jaqque.sbih.org/kplug/apt-pinning.html).

@elafargue If you're interested in helping out with hamradio packages in Debian, contact debian-hams@lists.debian.org and I'm sure we can get you started. (:

dranch, 2016-04-29:
Would be good to get a backport done for Direwolf though 1.3 is still in a beta phase as it's still on 1.3-dev-I.  I've found it's pretty stable though.

--David

irl, 2016-05-29:
Version 1.3 has now been released, doing updates on the package now and will attempt a backport once 1.3 is in testing (at most expecting a 10 day delay before I can do a backport upload, no less than 5 though to allow automated QA tests to run).

wb2osz, 2017-05-10:
A new stable version, 1.4, is now available.
I'm open to suggestions on how to make the packaging easier.

irl, 2017-05-11:
Updating the package to 1.4 now. A backport will be available for stretch once stretch has been released. As Debian is currently in a freeze, the package will be uploaded to experimental, not unstable. It will be reuploaded to unstable after the release.

chrisdebian, 2025-03-28:
[This](https://repology.org/project/direwolf/badges) is a helpful quick reference to the current versions.

Chris

dranch, 2025-03-28:
@chrisdebian - your post is off topic.  As mentioned in this discussion, Direwolf is already packaged by Debian people and tracking those versions is for other communities if they care to look.