wb2osz/direwolf #16: How to configure hamlib for ptt

stephanellis, 2016-01-13:
I saw in the December pre release that support for hamlib was added.  I'd like to give it a whirl, but I don't know how to configure it?  What shall I put on the PTT line?

Thanks!

stephanellis, 2016-01-15:
Using "PTT RIG /dev/ttyUSB0 122" gleaned from source, but it complains can't find the rig number.  I have verified using the hamlib tools that it does key the radio, but can't get direwolf to use it.  Thanks a bunch, this modem is decoding like a champ.  Now I just need it to key the radio.

wb2osz, 2016-05-03:
The hamlib interface code was updated and documented better in version 1.3.   Look in the User Guide.   Currently Section 9.2.8.4 "PTT Using hamlib."

"PTT RIG /dev/ttyUSB0 122"  has the last two parameters reversed.  Try  "PTT RIG 122 /dev/ttyUSB0"

closed 2016-05-03 by wb2osz

gerryk, 2016-09-14:
I am trying to use hamlib with Windows 1.3 build. Is hamlib enabled with this? I use a PTT specifier as follows: PTT RIG 135 COM7, but get the following message: Config file line 205: PTT with RIG is only available when hamlib support is enabled.

I guess hamlib support needs to be enabled?

Thanks
Gerry

dranch, 2016-09-14:
Do you have hamlib already installed?  That package is NOT included with Direwolf and needs to be installed from https://sourceforge.net/projects/hamlib/?source=directory

gerryk, 2016-09-14:
Hi David... I do. It's installed, in the path and accessible, but I get the "PTT with RIG is only available when hamlib support is enabled" message all the same.
Is the prebuilt win32 binary built with USE_HAMLIB defined?
/ Gerry

gerryk, 2016-09-15:
From config.c:

#ifdef USE_HAMLIB
.
.
#else
	      text_color_set(DW_COLOR_ERROR);
	      dw_printf ("Config file line %d: %s with RIG is only available when hamlib support is enabled.\n", line, otname);
#endif

So, I guess it's not...

gerryk, 2016-09-15:
I added the following to Makefile.win, added the hamlib includes to the search path and added libhamlib.dll.a to the MinGW library path and rebuilt...
------
CFLAGS += -DUSE_HAMLIB -I../../hamlib-master/hamlib-master/include
LDFLAGS := -lhamlib
-----
When I run, initially it looks good, then after a bit, it crashes...
-----
C:\MinGW>cd \Users\gerryk\Downloads\direwolf-master\direwolf-master

C:\Users\gerryk\Downloads\direwolf-master\direwolf-master>direwolf.exe
Dire Wolf version 1.3
**Includes optional support for:  hamlib**

Reading config file direwolf.conf
Available audio input devices for receive (*=selected):
 *  0: Microphone (2- USB Audio CODEC    (channel 0)
    1: Microphone (Realtek High Defini
    2: Microphone (USB Audio CODEC )
Available audio output devices for transmit (*=selected):
    0: Speakers (Realtek High Definiti
    1: Realtek Digital Output (Realtek
 *  2: Speakers (2- USB Audio CODEC )   (channel 0)
    3: Speakers (USB Audio CODEC )
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate.
**Hamlib Rig open error -14: Communication bus collision**
Ready to accept KISS client application on port 8001 ...
Ready to accept AGW client application 0 on port 8000 ...

C:\Users\gerryk\Downloads\direwolf-master\direwolf-master>
-----
Progress... of sorts.
/ Gerry