wb2osz/direwolf #17: OS X packaging

vielmetti, 2016-01-16:
The "Version 1.3 -- Development snapshot I -- December 2015" version says "Support for Mac OS X."  The README does not mention OS X but the Makefile.macosx looks like it's the place to start to build.

If I figure out the minimum edit to the README I'll submit a PR.

mistermatt2u, 2016-01-20:
Were you able to get 1.3 dev snapshot I to compile on your Mac? I had problems with it on my macbook air. I made some changes to the osx makefile and submitted a pull request. However, it is still pending.

wb2osz, 2016-05-03:
The User Guide chapter 6  (currently pages 23 -26) discusses Mac OS X.   I don't have a Mac so I depend on help from the user community for porting, documentation, and helping others.

closed 2016-05-03 by wb2osz