wb2osz/direwolf #19: Two direwolf stations transmit at same time (collide) how to adjust hold-off timer?

w8vfr, 2016-01-23:
I have two direwolf station sin close proximity. When each hears a position packet that is requesting digi service, each respond to the request at the same time colliding with one another. Is there a fix for this is direwolf.conf ?

dranch, 2016-01-23:
If you really have two digis in the same coverage area, the best solution is to DISABLE one of them.  If they have different coverage areas, you could consider setting up filters on both of them but them the whole point of digipeating APRS beacons becomes broken

w8vfr, 2016-01-23:
Yep, though of disabling one, but then the coverage goes away:/ 
Some means in the code has to hold the transmitter from keying until the frequency is clear. We just need a way to randomize this or manually set.

wb2osz, 2016-01-23:
That behavior is intentional.

For most transmissions, there is a random delay after the channel is clear to avoid collisions.
Digipeated frames are handled differently; they are transmitted as soon as the channel is clear. 
Google for aprs fratricide to see some discussion about this topic.

If you really want to change it to behave in a non-standard way, look for the following line in "digipeater.c" and change TQ_PRIO_0_HI to TQ_PRIO_1_LO.

	        tq_append (to_chan, TQ_PRIO_0_HI, result);

closed 2016-01-23 by wb2osz