wb2osz/direwolf #25: Direwolf talking to TNCs

JohnNoden, 2016-03-24:
Firstly may I say that I think this project is really well designed. I have a lot of experience in low-level software design so I very much appreciate the diagnostic facilities built in. I am a memeber of Raynet one of the emergency groups. We use APRS a lot and often the operators are not experienced with digital. I except that modern sound-card TNS can out perform traditional ones but only when they are set up correctly. Having the ability to talk to the TNCs in the Kenwood boxes etc. means that the system is plug and go. How much work would it take to make the upper levels talk to a TNC instead of the sound card thereby leaving the packet diagnostics, digipeater, iGate etc. in place?

Many thanks John - G8IOK

nayrnet, 2020-06-05:
You'll find Direwolf's TNC is far superior to any hardware TNC you can obtain, sure your existing TNC might work fine but DW is better and can decode more far more packets.. such a setup would be a downgrade in capabilities and performance.. like buying a sports car and swapping a lawn mower motor into it.

K2IE, 2020-06-05:
@nayrnet No doubt, but I own the TNC and would need to pull it and add a Rim Lite or such to the Alinco...so looking for a transtional approach for the time being.  73

wb2osz, 2025-03-30:
The "dev" development branch now has the ability to map a virtual channel to a network TCP KISS TNC.
See  https://github.com/wb2osz/direwolf-doc/blob/main/APRS-LoRa-VHF-APRS-Bridge.pdf 
This is mostly to allow a bridge between traditional APRS and new approaches such as LoRa APRS.
Currently, it allows only a TCP connection.  There is deman for a serial port connection as well so that will probably happen.
Obviously, no promises.  This is a hobby time project.