wb2osz/direwolf #28: Help with GPIO Raspberry

cybersyx, 2016-04-23:
Hi, this software is awesome and work very well but i've a little problem with PTT: I've a yaesu radio and open and close ptt with a simple relay. If i do gpio mode 1 out and gpio write 1 1 and gpio write 1 0 the ptt works perfectly but if i set on direwolf.conf PTT GPIO 1 the ptt doen't work. Thanks a lot for the help !

dranch, 2016-04-23:
You need to give more information for people to help you.  

Per the Direwolf Raspberry-Pi-APRS.pdf documentation (search for the term GPIO), the selection of a GPIO pin to use first depends on which Rpi version.  Which model and version do you have?  Next, which GPIO pin are you trying to use?  Please note that the header pin number does NOT reflect actual GPIO pin number according to Linux.  It's also worth mentioning that certain GPIO pins get initialized differently by Linux and depending how you wire things, if the RPI is shutdown, you might have your radio keyed up 100% of the time.  Test for this condition!   For me, I'm using 

    Rpi pin37(GPIO 26) to RS232 DB9-pin7(RTS)
    Rpi pin39(GND)     to RS232 DB9-pin5(GND)

Finally, I think the use of a 3.3v relay will add unnecessary delay compared to a simple transistor circuit.  Any specific reason to use a relay?

--David
KI6ZHD

cybersyx, 2016-04-23:
Thanks for the reply. Do u using the radio connected to the rpi with serial interface ?

dranch, 2016-04-24:
serial interface for PTT?  I've used an external USB to serial device for PTT, yes.  Maybe you mean the Rpi's internal serial port?  If so, no, I haven't tried to use that port for PTT.

--David
KI6ZHD

wb2osz, 2017-06-18:
Closing because the problem description lacked enough detail to understand the issue.  No further discussion here for over a year.

closed 2017-06-18 by wb2osz