wb2osz/direwolf #29: Strange CPU usage issue on Raspberry Pi

AngusF22, 2016-04-27:
I'm having an odd issue with Direwolf on the latest version of Raspbian. It's freshly compiled, and seems to work quite well, although the CPU usage seems very high under certain conditions. My preferred usage for the time being is as a home station connected via APRS servers only, for the time being. My Raspberry Pi is headless, and runs continuously for a number of home server tasks.

When I have ADEVICE set to null null, it will saturate one core of my Raspberry Pi's processor. If I comment out the ADEVICE null null line, it will refuse to start without an audio device attached, stating that it's pointless to continue. I respectfully disagree.

While ADEVICE is set to null null, htop shows two threads for Direwolf, each consuming 100% CPU.

If I install Pulseaudio, and comment out ADEVICE null null, leaving no audio device specified, Direwolf will happily begin running with much lower CPU usage. Pulseaudio will start two threads, each of which consumes a bit of CPU, and Direwolf will start three, each of which consumes about 15-20%.

Is Direwolf using an audio thread for its CPU throttling? That doesn't seem ideal.

Thanks!
Angus

wb2osz, 2016-07-04:
Dire Wolf uses many threads for the various concurrent activities.
Each audio input device has its own thread which reads data as quickly as the source can provide it.
In this case, the null device provides zero bytes at an unlimited rate and one CPU core is completely consumed.

Longer term, there should be an option to allow for this case.
(Maybe if it sees “null” for the audio input device, we won’t bother to open the device and start processing.)

For now, the work-around is to specify stdin or a UDP port as the audio source and don’t feed anything into it.
The thread for decoding that audio input channel will just sit there waiting because there is nothing to process.

chrisdebian, 2025-03-28:
Hi, @AngusF22 , do you know whether this is still an issue with the current version?

Chris

AngusF22, 2025-03-29:
I'm sorry, I can't say. I've switched to a different APRS setup.

wb2osz, 2025-03-29:
If you don't want audio input, set audio input stream to either udp or stdin.
The receive part will just sit there waiting and not consume any CPU time.
Closed.

closed 2025-03-29 by wb2osz