wb2osz/direwolf #33: Direwolf crash after receiving KISS frame

SARTrack, 2016-05-10:
I got my SARTrack software working with the Direwolf TCP/KISS interface on receive, but now I have a problem that Direwolf crashes when I transmit a KISS frame to it, my beacon.
I attach a screenshot of Direwolf in debug mode.
Direwolf DOES decode the KISS frame correctly, but immediately afterward crashes.

The KISS frame I send is identical to what I send to any TNC in KISS mode without any problems,

Beacon: ZL4FOX-5>APSAR,WIDE2-2:=1702.61S/14521.00E-/A=000272SARTrack Limited www.sartrack.co.nz

KISS Hex:
C0 00 82 A0 A6 82 A4 40 60 B4 98 68 8C 9E B0 6A AE 92 88 8A 64 40 65 03 F0 3D 31 37 30 32 2E 36 31 53 2F 31 34 35 32 31 2E 30 30 45 2D 2F 41 3D 30 30 30 32 37 32 53 41 52 54 72 61 63 6B 20 4C 69 6D 69 74 65 64 20 77 77 77 2E 73 61 72 74 72 61 63 6B 2E 63 6F 2E 6E 7A 5B 3A 42 61 72 74 73 20 44 65 76 65 6C 6F 70 6D 65 6E 74 20 50 43 C0

Bart Kindt
SARTrack developer
http://www.sartrack.co.nz/

[image]

SARTrack, 2016-05-11:
The problem persists regardless how simple I the make the data:

[image]
Packet:
ZL4FOX>APSAR:>Status 

Hex: 
C0 00 82 A0 A6 82 A4 40 60 B4 98 68 8C 9E B0 61 03 F0 3E 53 74 61 74 75 73 C0

SARTrack, 2016-05-11:
I also took of the Soundcard system, and switched to a TNC-X on a COMport.
This made no difference, so the crash is not caused by the Sound card interface...

PS: Using Windows 8.1.

Bart

wb2osz, 2016-07-03:
There was an issue with version 1.3 on Windows crashing when attempting to transmit and other erratic behavior.   This should be fixed in version 1.3.1.   This only affected the Windows version, not Linux.

closed 2016-07-03 by wb2osz