wb2osz/direwolf #35: Feature request WX info

samsonpwnz, 2016-05-25:
I have been trying to come up with a workaround for this myself, but I am at a standstill.
Is there any way to grab WX info from a file that is already properly formatted to send out on APRS and beacon using that? Or any combination of ways to get direwolf to beacon WX info. Greatly appreciated!

wb2osz, 2016-07-04:
Did you try using the custom beacon (CBEACON) type?    The INFOCMD option allows you to specify a command which generates the information part of the packet.

morcant, 2017-02-13:
I am also looking to report weather data. I have attempted using CBEACON with symbol="_" and an info command that returns a formatting temperature string, but no luck. Any examples?

wb2osz, 2017-06-14:
An example has been added to the User Guide.

Send a weather report with position: 

1.	Some other application periodically creates a file, called wxnow.txt, in the standard format.  (See    http://wiki.sandaysoft.com/a/Wxnow.txt )   This is the same format used in the APRS weather report packet.
2.	Configure a position beacon which inserts the last line, from the file, into the comment field.  In this particular case, it is very important that you don’t use COMMENT, POWER, HEIGHT, GAIN, FREQ, OFFSET, or any other options that would put anything else in the comment part.   The weather report needs to be in a very specific format.

PBEACON  LAT=42^37.14N  LONG=71^20.83W  SYMBOL=”weather station” COMMENTCMD=”tail -1 wxnow.txt”          
	
The result would be:

WB2OSZ-15>APDW15:!4237.14N/07120.83W_272/010g006t069r010p030P020h61b10150

As a sanity check, we run it through the decode_aprs utility.

$ echo 'WB2OSZ-15>APDW15:!4237.14N/07120.83W_272/010g006t069r010p030P020h61b10150' | decode_aprs

WB2OSZ-15>APDW15:!4237.14N/07120.83W_272/010g006t069r010p030P020h61b10150
Weather Report, WEATHER Station (blue), DireWolf, WB2OSZ
N 42 37.1400, W 071 20.8300
wind 11.5 mph, direction 272, gust 6, temperature 69, rain 0.10 in last hour, rain 0.30 in last 24 hours, rain 0.20 since midnight, humidity 61, barometer 29.98, ""

closed 2017-06-14 by wb2osz