wb2osz/direwolf #39: Suggest a fixed name for logging

bradanlane, 2016-05-31:
I've gone ahead and changed log.c so the logging file is a fixed name (traffic.log) rather than a log with a time stamp in it. The reason is to leverage logrotate to manage logs.

I'm not suggesting that "traffic.log" is the best name but the use of logrotate is a nice standard solution for Linux systems. I'm not aware of the Windows equivalent for logrotate.

bradanlane, 2016-06-23:
FYI: I've coded this and can submit if desired. It uses "-o" as a command line argument or "LOGNAME" as a conf file entry. Each take a file name. When a fie name is provided, activity logging will stay with the named file rather than change with the date.

The assumption is this option is used in conjunction with logrotate. When using logrotate, the `copytruncate` option is used.

wb2osz, 2017-06-18:
Similar functionality is now in the "dev" branch.
52927ce54ab419a67730522d06250569ddf8cf5c

closed 2017-06-18 by wb2osz