wb2osz/direwolf #41: Addition to SDR iGate Doc

IPingUPing, 2016-06-05:
Followed the instructions and received an error related to accessibility of the dongle.  Troubleshooting determined that blacklisting the current dvb drivers resolved.  May want to add something similar to this step:

Blacklist the RTL Dongle from being used as an actual TV Tuner
Open the blacklist file for editing (it may be empty)
sudo nano /etc/modprobe.d/raspi-blacklist.conf 
Add to the file on a new line
blacklist dvb_usb_rtl28xxu

bradanlane, 2016-06-21:
if we need to blacklist the SDR, should we use what other projects use - eg:

blacklist dvb_usb_rtl28xxu
blacklist dvb_usb_v2
blacklist rtl_2830
blacklist rtl_2832
blacklist r820t

wb2osz, 2016-07-04:
This was added to the documentation in the "dev" branch and will appear in the next release.

closed 2016-07-04 by wb2osz