wb2osz/direwolf #43: "Audio input device 0 error: No such device" after several minutes of usage

kevinhooke, 2016-06-09:
I'm running on a Raspberry Pi3 running Raspbian Jessie, pulled latest Direwolf from source. Using a SignaLinkUSB connected to the Pi. 
All is working ok, until after a few minutes of successful sending and receiving packets (using axcall) I get this error:
Audio input device 0 error: No such device
repeats 11 times then Direwolf exits.

I've seen this error running on a Pi1 with Wheezy too (more often), and just updated it to Jessie to see if latest Raspbian version fixed the issue. I haven't seen the issue reoccur on the P1+Jessie since I updated, but I did just see it on the Pi3+Jessie.

dranch, 2016-06-09:
Look at the various logs in /var/log/ such as messages, syslog or kern.log as I suspect a power issue.  How are you powering your Rpi?  It should be with at least a 5.1v @ 1A supply.  Next, how are you connecting the Signalink to the Pi?  Directly or via a hub?  If via a hub, Direwolf has had some known issues here so you might want to go direct but I would recommend to first try with a POWERED usb hub.  After that, I would suspect RFI messing with your RPi.  Does your USB cables have ferrites on each end?  What about the audio cables going into your radio?   How close is the Rpi to the TX antenna?

--David

kevinhooke, 2016-06-09:
Thanks for the tips David - I see in /var/log/messages:

```
Jun  9 00:28:39 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Thu Jun  9 00:29:09 2016 [try http://www.rsyslog.com/e/2007 ]
Jun  9 00:36:05 raspberrypi kernel: [ 1353.518448] usb 1-1.2: USB disconnect, device number 4
Jun  9 00:36:05 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Thu Jun  9 00:36:35 2016 [try http://www.rsyslog.com/e/2007 ]
Jun  9 00:36:06 raspberrypi kernel: [ 1353.856633] usb 1-1.2: new full-speed USB device number 5 using dwc_otg
Jun  9 00:36:06 raspberrypi kernel: [ 1353.979268] usb 1-1.2: New USB device found, idVendor=08bb, idProduct=2904
Jun  9 00:36:06 raspberrypi kernel: [ 1353.979290] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun  9 00:36:06 raspberrypi kernel: [ 1353.979303] usb 1-1.2: Product: USB Audio CODEC
Jun  9 00:36:06 raspberrypi kernel: [ 1353.979315] usb 1-1.2: Manufacturer: Burr-Brown from TI
Jun  9 00:36:06 raspberrypi kernel: [ 1354.000603] input: Burr-Brown from TI               USB Audio CODEC  as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.3/0003:08BB:2904.0002/input/input2

```
So looks like my Signalink disconnected and then reconnected again?
The Pi is a Pi3, and it has the 7" screen attached. It's running on two Anker power packs, one for the screen, one for the Pi. I think their rated output is 3A, but the Signalink is directly connected and not via a powered uhb, so I'll try that next. 

Re, RF, I'm using a Wuoxun HT, set to low power for testing.

A lot of things to try, but from seeing the logs this doesn't look like a Direwolf issue, so ticket can be closed and I'll do some followup on the power issues.

wb2osz, 2016-07-04:
Previous comment said this could be closed.

closed 2016-07-04 by wb2osz

bi7jta, 2016-11-04:
HI: I catch the same error, not low power cause, it is a bug in direwolf code.  When open [DIGIPEATER] or [IGTXVIA] mode,  large number  PBEACON will make USB SoundCard or System sounccard die! Because the direwolf application not  time interval set to TX to RF through SoundCard.
Solve the problem by:
1) Modify direwolf source code ,add  time interval to deal TX bean ; OR
2) Modify direwolf.conf setting: IGTXLIMIT 6 10 to IGTXLIMIT 1 1 (One BEAN TX per 1 minute)

Now i use the 2th, it work OK!

DE BI7JTA@gmail.com Twitter: bi7jta QQ Group : 290347330

dranch, 2016-11-04:
Hello BI7JTA,
Thanks for the report.  Could you post a diff of what's working for you so we can look at it?
--David
KI6ZHD