wb2osz/direwolf #44: PTT always on with serial port

cx5cba, 2016-06-25:
I use serial port to PTT in Windows 7 (Native RS232 and USB to RS232 adapter). When I turn on the Direwolf the TRX turn in TX.
I only made one change in the direwolf.conf:  "PTT COM1 RTS"

I tried the AGWPE and the PTT work fine

Thanks for help

dranch, 2016-06-26:
Hello CX5CBA,

Assuming the RTS signal is the correct signal, try "PTT COM1 -RTS".  Notice the added "-" which reverses the pin's logic.

--David
KI6ZHD

cx5cba, 2016-06-26:
Hello David,

I tried that you said but still the error.

Thanks

Carlos
CX5CBA

dranch, 2016-06-26:
Hmm.. that's very strange.  AGW is configured for the RTS signal on that 
COM port?

--David

On 06/26/2016 11:26 AM, cx5cba wrote:
>
> Hello David,
>
> I tried that you said but still the error.
>
> Thanks
>

cx5cba, 2016-06-26:
Yes, AGW use RTS for left channel and DTR for right channel. It´s very strange.

dranch, 2016-06-26:
Oh.. that's an entirely DIFFERENT config.  You'd want something like the 
following but make sure each PTT entry is in the correct CHANNEL section:

CHANNEL 0
PTT COM1 RTS
CHANNEL 1
PTT COM1 DTR

You might have to change RTS / -RTS / DTR / -DTR to match the right 
polarity of your PTT circuit to the radio.

--David
KI6ZHD

DwaineGarden, 2017-07-23:
I have the same problem.  When using the PTT with USB serial cable the radio is in TX all the time.  I think something has broken because it use to work fine.

DwaineGarden, 2017-07-23:
It's happening on both raspberrypi and a regular Linux computer.

dranch, 2017-07-24:
This ticket is over a year old and Direwolf 1.4 has been released with many fixes.  Dwaine, in your environment, which serial line are you using for PTT?  You should be using RTS (don't use DTR).

cx5cba, 2017-07-24:
"This ticket is over a year old and Direwolf 1.4 has been released with many fixes. Dwaine, in your environment, which serial line are you using for PTT? You should be using RTS (don't use DTR)."

Hello dranch.
I use v1.4 and still the PTT error only in Direwolf.

DwaineGarden, 2017-07-25:
I'm using -RTS on the dev branch.  I also tried two different USB serial cables on the raspberry Pi.  Same constantly open TX.  It use to work fine with TX on then TX off after the transmit.

dranch, 2017-07-25:
Please join and post about your issue on direwolf_packet@yahoogroups.com and we'll get your issue solved.  Github's "issues" section is for real bugs/enhancements so let's prove / disprove it there as too many people are successfully using Direwolf w/o issue.

cx5cba, 2017-07-25:
Dear David,

I successfully using Direwolf with native serial ports and USB adapters but not for FT-991 USB port, Therefore it´s a real issue for me.
Thanks for your cooperation

DwaineGarden, 2017-07-26:
Ok. I will post to the forum.

wb2osz, 2025-03-30:
When Windows reboots, it sets serial port RTS to the high state.  
I don't know if there is some way to tell the operating system to make the default low state.

closed 2025-03-30 by wb2osz