wb2osz/direwolf #46: config errors to stdout

bradanlane, 2016-07-05:
Currently most errors are sent to stdout. It would see to be easier to diagnose if errors (config and runtime) went to stderr. 

Is there a historical or functional reason for what goes to stdout vs stderr?

chrisdebian, 2025-03-28:
@bradanlane , you make a good point, do you know whether this is still the situation?

Chris

bradanlane, 2025-03-28:
I have not used direwolf since about 2018 so I can not speak to code enhancements over the past 8 years.

chrisdebian, 2025-03-28:
Thanks, @bradanlane . Looks like I don't have permission to close this.

dranch, 2025-03-28:
Why are we seeing a rash of ticket discussion updates from topics back in 2016?!

wb2osz, 2025-03-29:
Almost a decade old.
Closing.
Reopen if you think something still needs to be addressed.

closed 2025-03-29 by wb2osz