wb2osz/direwolf #48: Add possibility to define "one" log file name

aw-tech, 2016-07-22:
iIt's not issue ;)
Add possibility to define "one" log file name .
i'm use log file to display data on lcd .. on orangepi.
And need to check directory for new log files.
`//strftime (fname, sizeof(fname), "%Y-%m-%d.log", &tm);`
`strftime (fname, sizeof(fname), direwolf.log", &tm);`
change, in log.c  do the job.,
But it will be cool if we can set up it from command line ,
or maybe i missed something .,
Btw. Great job ...

dranch, 2016-07-24:
I would agree that log rotation on Linux would be better done with tools like "logrotate" but I don't think Windows has any facility like this.  I wonder if it would be wise to have different configurations for log files depending on the base OS (linux vs. Windows).  I don't know what OSX does here
--David
KI6ZHD

bradanlane, 2016-07-24:
My solution was to code this as a command line option and/or a CONF file option. The default is to continue to use date-stamped file name.

dranch, 2016-07-24:
The Direwolf User-Guide.pdf file mentions the LOGDIR directive (not mentioned in the stock direwolf.conf file -- it probably should) but it still uses the yyyy-mm-dd.log format.   Brandan, if you have some code that allows for a .conf / command line, option, could you consider creating a fork, commit your code there, and submit a Git pull back to John for inclusion in Direwolf?  I'm happy to help you do this.. it's not too hard.

aw-tech, 2016-07-25:
Hi,i made a fork , and add custom log file name , via command line -f.
I'm not programmer , just made some copy and paste..
For example : `sudo direwolf -t 0 -a 100 -d i -l /home/pi/direwolf -f mylog.log`
Without -f , standart log file name with date.
https://github.com/aw-tech/direwolf

dranch, 2016-07-25:
Ok, your 95% of the way there.  On this new fork you created wit hthe new commit, there is the "pull request" button (not the greatest terminology bit that's what Git calls it).  Click on that button to submit the proposed to John for inclusion in the Dev branch )if he accepts it).

aw-tech, 2016-07-25:
Ok thanks, i will try.

wb2osz, 2017-06-18:
Similar functionality is now in the "dev" branch.
52927ce54ab419a67730522d06250569ddf8cf5c

closed 2017-06-18 by wb2osz