wb2osz/direwolf #50: Crash when using raw KISS TNC

vvanders, 2016-08-01:
I have custom protocol that I'm working on(https://github.com/vvanders/nbplink) that I was hoping to use Direwolf as a raw KISS TNC for Signal Link and other soundcard based radio interfaces.

With the current version(1.3) Direwolf is crashing for me with the first packet sent across. I have packets successfully working on a Kenwood D710 and Mobilinkd via serial ports so I'm somewhat sure that the KISS framing is valid.

I'd be happy to dig into Direwolf more but I can't seem to build it for my platform(Win10) a "make direwolf" from cygwin w/ MinGW tools on the path returns:

```
$ make direwolf
gcc -Wall -Ofast -msse -Iregex -Iutm -Igeotranz -mthreads -DUSE_REGEX_STATIC -g   -c -o beacon.o beacon.c
beacon.c:68:12: error: conflicting types for 'localtime_r'
 struct tm *localtime_r(time_t *clock, struct tm *res)
            ^
In file included from beacon.c:40:0:
C:/msys64/mingw64/x86_64-w64-mingw32/include/time.h:269:34: note: previous definition of 'localtime_r' was here
 __forceinline struct tm *__cdecl localtime_r(const time_t *_Time, struct tm *_Tm) {
                                  ^
```

Anyway, repro is pretty simple, Win10, v1.3 I'm sending the following bytes over the TCP port:
```
Wrote [192, 0, 20, 163, 178, 115, 107, 92, 77, 140, 0, 0, 0, 0, 107, 92, 77, 140, 0, 0, 0, 0, 32, 116, 101, 115, 116, 10, 140, 153, 192]
```

direwolf.conf is:
```
#############################################################
#                                                           #
#               Configuration file for Dire Wolf            #
#                                                           #
#                   Windows version                         #
#                                                           #
#############################################################
#
# Consult the User Guide for more details on configuration options.
#
#
# These are the most likely settings you might change:
#
#	(1)   	MYCALL 	-  call sign and SSID for your station.
#
#			Look for lines starting with MYCALL and 
#			change NOCALL to your own.
#
#	(2)	PBEACON	-  enable position beaconing.
#
#			Look for lines starting with PBEACON and 
#			modify for your call, location, etc.
#
#	(3)	DIGIPEATER  -  configure digipeating rules.
#
#			Look for lines starting with DIGIPEATER.
#			Most people will probably use the given example.
#			Just remove the "#" from the start of the line
#			to enable it.
#
#	(4)	IGSERVER, IGLOGIN  - IGate server and login
#
#			Configure an IGate client to relay messages between 
#			radio and internet servers.
#
#
# The default location is "direwolf.conf" in the current working directory.
# An alternate configuration file location can be specified with the "-c" command line option.  
#
# As you probably guessed by now, # indicates a comment line.
#
# Remove the # at the beginning of a line if you want to use a sample
# configuration that is currently commented out.
#
# Commands are a keyword followed by parameters.
[... 420 lines cut ...]
# Advertise gateway position with beacon.

# OBEACON DELAY=0:15 EVERY=10:00 VIA=WIDE1-1 OBJNAME=WB2OSZ-tt SYMBOL=APRStt LAT=42^37.14N LONG=71^20.83W COMMENT="APRStt Gateway"  

# Sample speech responses.
# Default is Morse code "R" for received OK and "?" for all errors.

#TTERR  OK               SPEECH  Message Received.
#TTERR  D_MSG            SPEECH  D not implemented.
#TTERR  INTERNAL         SPEECH  Internal error.
#TTERR  MACRO_NOMATCH    SPEECH  No definition for digit sequence.
#TTERR  BAD_CHECKSUM     SPEECH  Bad checksum on call.
#TTERR  INVALID_CALL     SPEECH  Invalid callsign.
#TTERR  INVALID_OBJNAME  SPEECH  Invalid object name.
#TTERR  INVALID_SYMBOL   SPEECH  Invalid symbol.
#TTERR  INVALID_LOC      SPEECH  Invalid location.
#TTERR  NO_CALL          SPEECH  No call or object name.
#TTERR  SATSQ            SPEECH  Satellite square must be 4 digits.
#TTERR  SUFFIX_NO_CALL   SPEECH  Send full call before using suffix.
```

If you want to build and run nbplink I'm happy to provide detailed steps.

Thanks,
Val

dranch, 2016-08-01:
Hello Val,

You should send any trouble reports to the Direwolf email list at https://groups.yahoo.com/neo/groups/direwolf_packet and not use the Github's issue function.  Anyway, what version of Direwolf are seeing the crashes on?  1.3.0 was known to crash on Windows but 1.3.1 solved that.  What of Direwolf version are you seeing this compile issue with?  GIT/Master?  GIT/Dev?  What specific version of compiler on Windows10 (MinGW and GCC)?

--David

vvanders, 2016-08-01:
Thanks for the note, will do in the future(you may want to throw something up under a heading in Readme.md so more people see it).

You're right that 1.3.1 performs better, I had grabbed 1.3 and I no longer see the crash with 1.3.1. I'll go ahead and close this out and follow up if I run into anything else.

closed 2016-08-01 by vvanders