wb2osz/direwolf #51: Unrouted path digipeating?

sq8vps, 2016-08-05:
Hello.
Is there any way to run the digipeater for unrouted paths?

For example:
Frame transmitted by me:
`SQ8VPS>APRS,SP2-2:info`
And then frame repeated by digi:
`SQ8VPS>APRS,SP2-1:info`
And repeated by next digi:
`SQ8VPS>APRS,SP2*:info`

How can I do that?

73!

DL7AD, 2016-08-23:
I think there is no option for this. Is this even part of the APRS specification?

Regards Sven

sq8vps, 2016-08-23:
I don't know, but unrouted paths are used in many countries, for example
SS3-3, SP3-3, HU3-3 etc. Thank you for response.
73!

wb2osz, 2016-08-24:
Look in the User Guide, section 9.5.
There is an example which will digipeat the addresses MA3-3, MA2-1, etc.

DIGIPEAT 0 0 ^WIDE[3-7]-[1-7]$ ^WIDE[12]-[12]$|^MA[1-7]-[1-7]$

Take the example above and change MA to SP.

closed 2016-08-24 by wb2osz

sq8vps, 2016-08-24:
Thanks for response and your very good app!
73!