wb2osz/direwolf #52: Compile Fails under Fedora 24 with GCC 6.1

DwaineGarden, 2016-08-22:
When trying to compile direwolf under Fedora 24 with gcc v6.1 I get this output.

gcc -O3 -pthread -Igeotranz -ffast-math -DUSE_ALSA   -c -o log.o log.c
gcc -O3 -pthread -Igeotranz -ffast-math -DUSE_ALSA   -c -o telemetry.o telemetry.c
gcc -O3 -pthread -Igeotranz -ffast-math -DUSE_ALSA   -c -o dwgps.o dwgps.c
gcc -O3 -pthread -Igeotranz -ffast-math -DUSE_ALSA   -c -o dwgpsnmea.o dwgpsnmea.c
gcc -O3 -pthread -Igeotranz -ffast-math -DUSE_ALSA   -c -o dwgpsd.o dwgpsd.c
dwgpsd.c:82:26: error: storage size of ‘gpsdata’ isn’t known
 static struct gps_data_t gpsdata;
                          ^~~~~~~
<builtin>: recipe for target 'dwgpsd.o' failed
make: *** [dwgpsd.o] Error 1

dranch, 2016-08-22:
Do you have libgps-dev installed per the Direwolf APRS Tracker documentation?  If so, what version?

--David

wb2osz, 2016-08-23:
Fixed in dev branch.
commit b2548ec58f44f4b651626757a166b9f4f18d8000

closed 2016-08-23 by wb2osz