wb2osz/direwolf #53: Digipeating fails but Igate continues to function

GWatson55, 2016-09-02:
I installed v1.3.1 based on the text I read about the previous version having some Windows keying problems.

My Igate/digi is up an running but not without some other software help.

For some background, I'm running Windows 7 on a netbook (CPU useage ~ 2 - 5 %) and use a USB to serial adapter.  I have used two different adapters with, to the best of my knowledge, equal succes.  One uses a Prolific chipset and the other is FTDI.  Both use "legitimate" chipsets and have given me no trouble in other applications.

There are actually two problems I have encountered and they are as follows:

At first, I ran only as an Igate station.  Setup was straight forward and I was up and running in little time.  The problem was the Dire Wolf engine would abruptly terminate for no apparent reason.  The time-frame for the problem could be in as little as within the first hour to as many as 14 hours.

My work-around for this was to include a process monitor which would automatically restart Dire Wolf upon unexpected termination.  With this in place, I had no other issues with the engine and everything worked fine without having to check on it's status.  If the engine terminates, it is restarted within 20 seconds.  The process monitor also logs unexpected terminations.

The second problem arose when I implemented the digipeater functionality.  Setup was clean and fast due to the excellent documentation and my digi was on the air quickly.  Unfortunately, something happens where the engine fails to send the command to digipeat.  I see the command on the monitor window but the LED indicator on my interface (RTS > opto > PTT) does not indicate that the RTS signal changes states.

Simply restarting the DW engine corrects the problem until the next time this happens.  The problem is random and may occur within a very few minutes or may not occur again for a couple of hours.  My work-around for this problem was to schedule tasks to terminate the engine on a regular basis (once every 20 minutes).  Once terminated, the process monitor restarts the the engine and the station is up again within the minute.

At this point, the scheduled tasks and the process monitor keep the station on the air without human intervention but I still see times when the digipeater function fails prior to time for the scheduled task to purposely restart the engine.  I've minmised my digipeater downtime but something is still not right with v1.3.1.  Again, the Igate functionality continues without a hitch.

Is there anything I can do to collect more useful data for the developer?

73, Greg

dranch, 2016-09-02:
Hello Greg,

> My work-around for this was to include a process monitor which would
> automatically restart Dire Wolf upon unexpected termination. With this
> in place, I had no other issues with the engine and [...]

Curious, are you doing this process monitoring and restart using native 
Windows tools or some other 3rd party tools?  It might be helpful for 
some people if you post what you did here and maybe John can add it into 
the official documentation.

> The second problem arose when I implemented the digipeater
> functionality. Setup was clean and fast due to the excellent
> documentation and my digi was on the air quickly. Unfortunately,
> something [...]

You mentioned that you're capturing logs now.  When things crash and 
restart, are you seeing any specific errors?  To increase logging, per 
the User Guide - Section 9.14) you can add "-d o" which will show PTT / 
DCD signals.  There is "-d i" which will show I gate operations.  
Another issue you might be seeing is illegal characters or specific 
corrupt frames that are crashing Direwolf (it's happened in the past).  
As such, you can add "-d p" to log all packets.  Once a failure occurs, 
you might be able to determine if a local station's packets are freaking 
our Direwolf.

Beyond that, I'm not aware of any "verbose logging" feature in Direwolf.

Btw, in your Direwolf config, disable AGW and TCPKISS in your 
direwolf.conf file by setting the socket number to "0"

--David
KI6ZHD

GWatson55, 2016-09-03:
David,

Thanks for your reply and the information. I took some time before replying in order to try a couple of other things.  I will try the suggestions you made asap.

You're right about me using third party software, at least in part, to terminate and restart DW. Since I started off setting things up to use the Igate functionality, I started looking for something "canned" which would solve the problem of a crash while I was unable to be there and monitor the pc. I tried a couple of programs but settled on a one named "Restart On Crash". I found it to be the only one which started DW correctly after a crash. I'll add links at the bottom of this e-mail. I am using Windows to terminate the engine at regular intervals (every 20 minutes) by taking advantage of Task Scheduler. I went back, before replying, and attempted to use TS to restart DW thinking it would be great to have everything taken care of within Windows. Unfortunately, that didn't work well. When TS calls DireWolf.exe, it doesn't seem to "remember" where the rest of the support files are and DW does not start correctly. This is the same thing that happened with the crash monitor program I tried. I don't have a lot of experience with TS so it is probably something I have missed in the task setup. I did not try all of the programs mentioned in the link describing eight different restart type programs. I did try App Monitor with poor results and ROC. I'm not sure if some or all of these programs simply offer an easier way to deal with Task Scheduler because they apparently work with Windows versions prior to 7.

My antivirus software says ROC is clean but who knows. It does not need to be installed so I thought that was good as well. I found it on a blog and then took some time last night to try and research it's origins. I did find the author's name but not a direct contact method for him unless I overlooked it. I did see he had recently answered questions about his program so he's still around. There was nothing on the blog or within the software that states whether it's copyrighted or open source and that's what I was really interested in since it could disappear if the his blog went away. If it's open source, it would be nice to be able to possibly add it to the DW distro. At least that was my thought.

I would be happy to go back to the thread in the forum and explain what I have done but wanted to run it past someone else since it involves the third-party software. What do you think?  This is working for me but hate to mentions something which may or may not be available to download tomorrow.

I am also attaching a snippit of the txt file recorded by ROC. It doesn't offer much insight into what happened but does show that it is effective in the event of an unexpected termination of DW or, for that matter, whatever it's target is.  Any time listed that is not :05, :25 or :45 is a crash.

Here are the links that lead me to ROC:

https://www.raymond.cc/blog/download/did/1914/

http://w-shadow.com/blog/2009/03/04/restart-on-crash/

Original article

https://www.raymond.cc/blog/keep-application-running-by-automatically-rerun-when-closed/

I have one question but can post to the forum if that's better.  DW mentions several different packages it will work with as a "drop-in" replacement for AGWPE.  I have had no luck with with Winpack.  It just doesn't seem to see the DW engine.  Will it work and I'm just missing something?

I would love to help more if there is something I can do.

Thanks and 73,
Greg Watson    KA9DTZ

-- 
8/31/2016 6:05:01 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe is not running
8/31/2016 6:05:01 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe will be treated as crashed/hanged
8/31/2016 6:05:22 PM	Executing a command : "C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe"
8/31/2016 6:05:24 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe has recovered
8/31/2016 6:25:02 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe is not running
8/31/2016 6:25:02 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe will be treated as crashed/hanged
8/31/2016 6:25:23 PM	Executing a command : "C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe"
8/31/2016 6:25:24 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe has recovered
8/31/2016 6:45:01 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe is not running
8/31/2016 6:45:01 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe will be treated as crashed/hanged
8/31/2016 6:45:22 PM	Executing a command : "C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe"
8/31/2016 6:45:23 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe has recovered
8/31/2016 6:52:32 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe is not running
8/31/2016 6:52:32 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe will be treated as crashed/hanged
8/31/2016 6:52:53 PM	Executing a command : "C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe"
8/31/2016 6:52:54 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe has recovered
8/31/2016 7:05:01 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe is not running
8/31/2016 7:05:01 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe will be treated as crashed/hanged
8/31/2016 7:05:22 PM	Executing a command : "C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe"
[... cut ...]
9/2/2016 1:08:39 PM	Executing a command : "C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe"
9/2/2016 1:08:41 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe has recovered
9/2/2016 1:25:02 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe is not running
9/2/2016 1:25:02 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe will be treated as crashed/hanged
9/2/2016 1:25:23 PM	Executing a command : "C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe"
9/2/2016 1:25:25 PM	C:\Downloads 2016\Packet and APRS\Dire Wolf\direwolf.exe has recovered

dranch, 2016-09-03:
Hey Greg,

Are you on the Direwolf Yahoo group?  It would be better to have you 
re-post this very email there and I'll reply to it.  Ok?

--David
KI6ZHD

wb2osz, 2023-04-10:
No activity for almost 7 years.  Closing.

closed 2023-04-10 by wb2osz