wb2osz/direwolf #54: How does one generate an IGLOGIN passcode?

chrismetcalf, 2016-09-07:
I've scoured http://aprs2.net/ and can't seem to figure out how to sign up and generate a passcode. [The example config](https://github.com/wb2osz/direwolf/blob/master/sdr.conf#L23) recommends contacting the author if you can't figure out how. 😄 

Also, I'm dabbling in receive-only while studying for my exam, is there a way to relay to aprs2.net without having a callsign yet?

KB3EFS, 2016-09-07:
Chris,

A call sign _IS REQUIRED_ for access to aprs2.net as all traffic sent to
it is subject to transmission via ham radio at various points. Thus you
can not relay to aprs2.net until you have a call sign.

Once you have a call sign then creating a passcode becomes very easy.

Dave
KB3EFS

chrismetcalf, 2016-09-07:
Thanks @KB3EFS! I'll get cranking on that license then.

Is there documentation somewhere on how to create a passcode once I've got a callsign?

KB3EFS, 2016-09-07:
Chris,

Yes, it would depend on your OS and APRS client chosen. Also note that
if your call sign changes due to systematic upgrade, relocation to
another call sign district, or selection of a vanity call sign, that
your passcode well also have to change.

My recommendation would be use of any of the flavors of Linux and use of
Xastir as your APRS client.

Dave
KB3EFS

twkisner, 2016-09-07:
That's not entirely accurate.  Your APRS passcode is a hash of your callsign and does not change unless your callsign changes.  It works across all applications and OSs.
The algorithm is not secret and there are webpages to generate your own passcode ( http://apps.magicbug.co.uk/passcode/ for example).
73,
Tom KN6Q
Date: Wed, 7 Sep 2016 10:02:44 -0700

KB3EFS, 2016-09-07:
Tom,

First Chris needs the license and associated call sign. Then after that
everything else falls into place.

Unfortunately some have faked call signs and gotten passcodes that are
now corrupting the system.

I was answering "Yes" to the question "Is documentation available" based
on my use of Ubuntu or Debian Linux and Xastir; I do not use other
operating systems.

73
Dave
KB3EFS

twkisner, 2016-09-07:
Dave-
You're quite correct that a license and call sign are absolutely required.  Chris, the tools are out there, but please don't use them until you have your ticket.
However, Chris is asking a legitimate question, I'd rather he get an honest answer than dance around it.  What good could it possibly do? There are at least a dozen web sites and applications that generate APRS passcodes, if this is a security risk for the APRS-IS infrastructure the cat is well out of the bag...
73,
Tom KN6Q

Date: Wed, 7 Sep 2016 10:40:53 -0700

wb2osz, 2017-01-22:
The official policy is that users should contact the IGate client authors to get the passcode.
Contact me directly at wb2osz//arrl//net and supply your ham radio callsign.

As others have pointed out, it doesn't take long to figure out how to generate it yourself.

closed 2017-01-22 by wb2osz