wb2osz/direwolf #55: Support Direwolf based pre-emphasis for radios using 9600BAUD connection

dranch, 2016-09-13:
Making this an official issue to not be forgotten and maybe implemented someday

More and more users (including myself) would like to leave our radios in "9600 baud mode" where the soundcard is connected  to the radio's discriminator output).  The issue here is that the resulting RF output would be flat audio.  This is BAD as AFSK1200 signals should have the high tone be significantly higher so that when the receiving radio hears the signal and applies it's expected de-emphasis, the resulting signal going to it's TNC should be flat.

I know that Svxlink for Linux (a Linux based repeater controller / Echolink node) supports this kind of functionality so it's technically possible.

--David

wa7nwp, 2016-09-13:
> More and more users (including myself) would like to leave our radios in "9600 baud mode" where the soundcard is connected to the radio's discriminator output). The issue here is that the resulting [...]

It's actually a mix with many stations today (Kenwoods) running flat
audio.   It would be nice if the decoding software was smart enough to
automatically compensate for audio twist if needed...

Another interesting science project would be to simply count how many
packets of each type were heard on the air.

Bill

dranch, 2016-09-13:
Hello Bill,

Adding comments to these tickets really belong in the Yahoo group.  
Anyway, John *did* write up a lot of behaviors he found out there in the 
real world.  See the 
"https://github.com/wb2osz/direwolf/blob/dev/doc/A-Better-APRS-Packet-Demodulator-Part-1-1200-baud.pdf" 
doc at:

    https://github.com/wb2osz/direwolf/tree/dev/doc

There are other excellent write ups that John wrote there that talk 
about his various testing and compensations that Direwolf does to 
support improper signals.

--David

wb2osz, 2020-03-21:
I would be hesitant to pre-distort the outgoing audio.  If it is not precisely matched to be the inverse of further distortions downstream, it's likely to create more problems than it solves.

closed 2020-03-21 by wb2osz