wb2osz/direwolf #58: Dire Wolf terminal displays wrong frequency

vk2byf, 2016-09-21:
In Australia we use 145.175MHz for APRS.
When I run rtl_fm -f 145.175M - | direwolf -c sdr.conf -r 24000 -D
on my RPi 2, (latest Jessie), the program runs but tells me it's tuned to 145427000Hz (252kHz high)
I'm new to Linux. I know enough to be dangerous but can't diagnose this myself.
Thanks, Bob, vk2byf

Manawyrm, 2016-12-12:
Hello Bob, 

this is normal. This message originates from rtl_fm and not from direwolf. 
rtl_fm automatically tunes a little bit off the specified frequency to make sure it can receive the data you want to receive. This has to do with the way I/Q-sampling works. (and also to prevent problems with the LO-frequency). 

Best wishes,
Tobias

wb2osz, 2017-05-10:
See explanation from Tobias.

closed 2017-05-10 by wb2osz