wb2osz/direwolf #60: GPIO PTT

W4JPA, 2016-09-29:
Do I have to install Hamlib to use a GPIO pin to PTT the radio (using an I/F board of course)?
I can directly trigger radio PTT with ~ "write 0 0" (for GPIO 17)
Although I have uncommented the "PTT GPIO 25" line in direwolf.conf and changed it to  0 for pin 17 It is not triggering when I use ~ "axcall w4jpa-1 ax0". Hmm!

dranch, 2016-09-29:
Using Git "issues" is not for getting support.  Please close this ticket, join the Yahoo Direwolf support group at  direwolf_packet@yahoogroups.com and I'll be happy to help you there.

W4JPA, 2016-09-29:
> <mailto:direwolf_packet@yahoogroups.com> and I'll be happy to help you

Sorry about that. These things can be confusing of your not suing them a lot

closed 2016-09-29 by W4JPA