wb2osz/direwolf #63: DireWOlf virtual TNC and ARDOP

sp2ong, 2016-10-29:
Hi,
It  will be fine that DireWolf will be have support new hamradio protocol ARDOP http://www.winlink.org/content/ardop_overview 
I know that exist ARDOP source by G8BPQ which implement in BPQ32 software https://github.com/g8bpq/LinBPQ/blob/master/ARDOP.c 

or use ARDOP version C by G8BPQ https://dl.dropboxusercontent.com/u/31910649/ARDOPC.txt

It maybe will easy or not :-) add ARDOP use via Sound card and with RMS Express or other Winlink application  like PAT Wilnik client: https://github.com/la5nta/pat

73 Waldek SP2ONG

dranch, 2016-10-29:
Just my $0.02 but Direwolf is an AX.25 TNC and APRS client/gateway.   ARDOP has no relation to packet or AX.25 and Ithink of it much more along the lines of Pactor.  As you mentioned, there is a Linux version of ARDOP now and the BPQ BBS software has API support for it which is a version of BPQ's SCS Pactor TNC support.  Other than that, as you mentioned, Pat *does* support ARDOP (though an older version).  Once ARDOP code development settles down, I have little doubt PAT will catch up support it.

wb2osz, 2017-06-18:
After looking over all the ARDOP documentation, I have pretty much the same reaction as dranch.
Dire Wolf is a software replacement for a traditional AX.25 Terminal Node Controller (TNC) box.   It has simple modems and knowledge of the AX.25 protocol and APRS.  
ARDOP is an entirely different animal.  From what I see, it is used to set up a reliable communication with one other station at a time.   So it is good for conversations or moving large amounts of data like AX.25 connected mode.  BBS or e-mail applications could use either one to move data reliably between two points.
I didn't notice any broadcast to anyone mode that would allow APRS to be layered on top of it.
In summary, I really don't see how they would fit together.

closed 2017-06-18 by wb2osz