wb2osz/direwolf #65: is possible to using USB audio/in/out cable

mikolaj24, 2016-11-09:
Hello,
I spossible to using cable USB<->2 jack ausio
not soundcard only directly from usb?

virtualdxs, 2016-11-09:
The USB device you are talking about *is* a soundcard. But theoretically,
if it weren't, no that would not be possible.

dranch, 2016-11-09:
Github is not a place to get Direwolf support.. it's a place for reporting bugs or feature enhancements.  Please subscribe to direwolf_packet@yahoogroups.com to vet your ideas first.  To your specific question, I bet your "USB device" is a sound card so depending on what you want to do, something might be possible but Direwolf today does not natively support the JACK or PulseAudio audio systems.  It only natively supports ALSA.

--David

mikolaj24, 2016-11-12:
I thinking about this cable http://www.banggood.com/Original-2-Pins-USB-Programming-Cable-for-BAOFENG-Walkie-Talkie-p-947095.html?rmmds=search
no sound modem.
 
W dniu 2016-11-09 18:53:50 użytkownik David Ranch <notifications@github.com> napisał:
Github is not a place to get Direwolf support.. it's a place for reporting bugs or feature enhancements. Please subscribe to direwolf_packet@yahoogroups.com to vet your ideas first. To your specific question, I bet your "USB device" is a sound card so depending on what you want to do, something might be possible but Direwolf today does not natively support the JACK or PulseAudio audio systems. It only natively supports ALSA.
--David

virtualdxs, 2016-11-12:
That is a programming cable. It will not work. Please get support elsewhere
and close this issue.

closed 2020-03-21 by wb2osz