wb2osz/direwolf #67: Feedback TCPKISS/AGWPE interfering with Linux AX25 stack and crashing connected mode

elafargue, 2016-11-24:
Just testing the current dev build (at d85abe214f006599dd84c31df7db4c1f292a6f06 ).

AGWPE support in connected mode didn't work out for me on a session using Outpost. Below is the log.

On that particular setup, direwolf is connected to the Linux AX.25 stack.

The issue is that when a session starts on the AWGPE socket, the packets are still forwarded on to the kiss TNC port and the Linux AX25 stack, which issues a series of "DM" right away, since the stack didn't initiate the connection. I also have the exact same issue when using TCPKISS and Winlink Express if the Linux AX25 stack is running.

One workaround is to avoid connecting direwolf to the Linux AX25 stack altogether, which is a bit brutal.

Another workaround would be to detect when a connected session is made through either the TCPKISS or AGWPE emulation port and avoid copying the packets to other ports during that time.

The third workaround, of course, is to use ldsped which sits behind the Linux AX25 stack and thus plays nice with it.

IMHO this is a generic issue with Direwolf, as soon as we go beyond UI frames. Thoughts?

```
Nov 24 18:27:10 Connected to AGW client application 0...
Nov 24 18:27:10 Ready to accept AGW client application 1 on port 8001 ...
Nov 24 18:27:11  Attempting connect to W3XSC-1 ...
Nov 24 18:27:12 [1L] W6ELA>W3XSC-1:(SABME cmd, p=1)
Nov 24 18:27:13 W3XSC-1 audio level = 77(22/40) [NONE]   ||||_____
Nov 24 18:27:13 [1.1] W3XSC-1>W6ELA:(DM res, f=1)
Nov 24 18:27:13 W3XSC-1 doesn't understand AX.25 v2.2. Trying v2.0 ...
Nov 24 18:27:14 [1L] W6ELA>W3XSC-1:(SABM cmd, p=1)
Nov 24 18:27:15 W3XSC-1 audio level = 74(23/39) [NONE]   |||||____
Nov 24 18:27:15 [1.2] W3XSC-1>W6ELA:(UA res, f=1)
Nov 24 18:27:15 Stream 0: Connected to W3XSC-1.  (v2.0)
Nov 24 18:27:17 W3XSC-1 audio level = 73(22/36) [NONE]   ||||||___
Nov 24 18:27:17 [1.2] W3XSC-1>W6ELA:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)[JNOS-2.0k.xsc.5-IHM$]<0x0d>You have 0 messages.<0x0d>(#0) ><0x0d>
Nov 24 18:27:17 [1L] W6ELA>W3XSC-1:(DM res, f=1)!<0xce>
Nov 24 18:27:18 [1L] W6ELA>W3XSC-1:(DM res, f=1)G<0xd3>
Nov 24 18:27:18 [1L] W6ELA>W3XSC-1:(RR res, n(r)=1, f=0)
Nov 24 18:27:18 [1L] W6ELA>W3XSC-1:(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)A XSCPERM<0x0d>
Nov 24 18:27:22 [1L] W6ELA>W3XSC-1:(RR cmd, n(r)=1, p=1)
Nov 24 18:27:23 W3XSC-1 audio level = 76(22/40) [NONE]   ||||||___
Nov 24 18:27:23 [1.2] W3XSC-1>W6ELA:(DM res, f=1)
Nov 24 18:27:23 Stream 0: AX.25 Protocol Error E: DM received in state 4.
Nov 24 18:27:23 Stream 0: Disconnected from W3XSC-1.
```
The same session but using the Linux AX25 stack and LDSPED which emulates AGWPE:

```
Nov 24 18:34:38 [1L] W6ELA>W3XSC-1:(SABM cmd, p=1) <0x10>
Nov 24 18:34:39 W3XSC-1 audio level = 74(23/39) [NONE]   ||||||___
Nov 24 18:34:39 [1.2] W3XSC-1>W6ELA:(UA res, f=1)
Nov 24 18:34:40 W3XSC-1 audio level = 73(23/36) [NONE]   ||||||___
Nov 24 18:34:40 [1.2] W3XSC-1>W6ELA:(I cmd, n(s)=0, n(r)=0, p=0, pid=0xf0)[JNOS-2.0k.xsc.5-IHM$]<0x0d>You have 0 messages.<0x0d>(#0) ><0x0d>
Nov 24 18:34:43 [1L] W6ELA>W3XSC-1:(I cmd, n(s)=0, n(r)=1, p=0, pid=0xf0)A XSCPERM<0x0d>7<
Nov 24 18:34:45 W3XSC-1 audio level = 74(23/39) [NONE]   ||||||___
Nov 24 18:34:45 [1.2] W3XSC-1>W6ELA:(I cmd, n(s)=1,n(r)=1, p=0, pid=0xf0)xscperm: 5 messages  -  0 new.<0x0d>(#1) ><0x0d>
Nov 24 18:34:46 [1L] W6ELA>W3XSC-1:(I cmd, n(s)=1,n(r)=2, p=0, pid=0xf0)A XSCEVENT<0x0d>
Nov 24 18:34:48 W3XSC-1 audio level = 74(23/38) [NONE]   |||||____
Nov 24 18:34:48 [1.2] W3XSC-1>W6ELA:(I cmd, n(s)=2,n(r)=2, p=0, pid=0xf0)Huh?<0x0d>(#1) ><0x0d>
Nov 24 18:34:49 [1L] W6ELA>W3XSC-1:(I cmd, n(s)=2,n(r)=3, p=0, pid=0xf0)L> XSCEVENT<0x0d>
Nov 24 18:34:52 W3XSC-1 audio level = 75(23/39) [NONE]   |||||____
Nov 24 18:34:52 [1.2] W3XSC-1>W6ELA:(I cmd, n(s)=3,n(r)=3, p=0, pid=0xf0)Mail area: xscperm<0x0d>5 messages  -  0 new<0x0d><0x0d>None to list.<0x0d>(#1) ><0x0d>
Nov 24 18:34:53 [1L] W6ELA>W3XSC-1:(I cmd, n(s)=3,n(r)=4, p=0, pid=0xf0)B<0x0d>
Nov 24 18:34:54 W3XSC-1 audio level = 74(23/38) [NONE]   |||||____
Nov 24 18:34:54 [1.2] W3XSC-1>W6ELA:(DISC cmd, p=1)
Nov 24 18:34:54 [1L] W6ELA>W3XSC-1:(UA res, f=1)
```

   Hoping this is helpful!

dranch, 2016-11-25:
Hello Ed,

Are you 100% sure that the CALLSIGN+SSID across Direwolf, Linux AX.25, and AGW/LDSPED are all unique?  They must be unique or you'll have stacks competing with eachother just like this.

--David
KI6ZHD

elafargue, 2016-11-25:
Great point, David. Using ldsped it's not a problem to use the same SSID,
since ldsped is using the Linux AX25 stack. You can see the trace above,
which works fine.

But it might be a good workaround when using the AGWPE emulation layer on
direwolf directly, and possibly with connected TCPKISS. I will test and
report.

elafargue, 2016-11-25:
Following up: yes, as per David's comment, the SSID used when going through the TCPKISS or AGWPE sockets has to be different from the SSID used for the interface on the Linux AX25 stack. Rookie mistake, eh? Might be worth mentioning in the documentation somewhere, better safe than sorry!

wb2osz, 2017-01-03:
So, did it work properly with only Outlook PM and Dire Wolf?

elafargue, 2017-01-03:
Thanks for the reminder: yes, by using a unique SSID between Direwolf and
Linux, the AGWPE emulation layer works perfectly with Outlook PM. Good job!

elafargue, 2017-03-30:
We can consider this closed, thanks

closed 2017-03-30 by elafargue