wb2osz/direwolf #68: Outernet support for the apes feed.

DwaineGarden, 2016-11-25:
Request to add Outernet support for their aprs feed.  It would be nice to see.

dranch, 2016-11-25:
Could you provide a URL that better explains this request?

DwaineGarden, 2016-11-26:
The L band sat for Outernet has an APRS feed.    The YAAC can grab the feed from the SDR hooked up to the satellite antenna.  Could be setup to natively have a IP port like we have for other protocols.
•Interface with applications by
◦AGW network protocol
◦KISS serial port
◦KISS network protocol
*Outernet satellite protocol

Here is the example from YAAC support.
http://www.ka2ddo.org/ka2ddo/YAACdocs/config_outernet.html

Details on the aprs feed.   
http://www.rtl-sdr.com/outernet-released-rxos-version-3-automatic-decompression-aprs-noaa-weather-data-news-updates/

Outernet fourm with details and discussion around APRS with the down feeds.
http://forums.outernet.is/t/aprs-txt-html-what-is-coming-down/2382/4