wb2osz/direwolf #70: Question about RAW packets from IF to Internet

kovalroma, 2016-12-08:
Hello!
I am not sure is it right place for my question but anyway. 

I've installed program. Here my station 
https://www.aprsdirect.com/details/main/sid/456583

According to statistics tab I've received 919 packet from UR5KSH.
According to RAW data I send only information about my position.  
"08.12.2016 05:53:10+01:00: UR3PHP-10>APDW13,qAR,UR3PHP-10:!5045.33NR02521.80E&RaspberryPi + RTL-SDR". 

But normally there should be some data from UR5KSH. Something like 
"08.12.2016 05:53:10+01:00: UR3PHP-10>APDW13,qAR,UR5KSH-10:!5045.33NR02521.80E&Digipeater". 

My question is. Is it a problem of aprsdirect site or it's a problem of my config file? 
Why aprs.fi site http://aprs.fi/info/a/UR3PHP-10 doesn't show UR5KSH station ? 
I am really confused. 

Sorry if this is not right place.

dranch, 2016-12-08:
Indeed.. the Github bug tracker is NOT the right place for support.  Please see https://groups.yahoo.com/neo/groups/direwolf_packet .  I encourage you to join that group and post your question there.  In that new thread, we can discuss http://aprs.fi/info/UR5KSH* which doesn't show the -10 SSID.

--David
KI6ZHD

kovalroma, 2016-12-08:
I will close topic here. 
I didn't know about yahoo group.
Thank you for understanding and sorry for mistake.

closed 2016-12-08 by kovalroma