wb2osz/direwolf #75: Way to change/add preamble for vox?

james-jimjam, 2017-01-03:
Hi,
I know and fully understand that Vox is not recommended by any means however I am having to build a small, cheap and easily replicated setup for a major event coming up in April. We will need to track our support and response vehicles around a large area of hilly countryside so plan to have a small network of digipeater's. We will be using Wouxun and Baofeng handheld radios that we already have for this and using direwolf on raspberry pi's. I have got everything running perfectly except the vox does not kick in in time for the first bit of data. Is there a way of increasing (I think there is a little bit) or adding some extra preamble (or tone) to the transmission so that it will key the vox in time for the data. 

I would much appreciate any help you can provide. I know my way around code so if you happen to just know where I can change a value then I can probably do that. 

Thanks in advance!
James, 2E0JPM

P.S. Direwolf is awesome.... I'm blown away by the features and abilities it has. Thank you so much!!

wb2osz, 2017-01-03:
This is the place for reporting software bugs and making enhancement requests.

It is not the place for getting support.

The main page lists a few discussion groups for asking questions and sharing information.

 

Increase the TXDELAY value as explained in the User Guide.

wb2osz, 2017-06-15:
Tip:   Avoid use of VOX built into your transceiver.

This is designed for voice operation and will keep the transmitter on about a half second after the transmit audio has ended.  This is much too long.  Others will probably start transmitting before you stop.

For an explanation, see the section called, “Radio Channel – Transmit Timing.”

closed 2017-06-15 by wb2osz