wb2osz/direwolf #90: Axspawn

felixdube, 2017-03-18:
Hi there,

I have notice an issue when using Direwolf along with Axspawn. Here is my setup. I have two computers communicating over an AX.25 link using Direwolf 1.3. There is one computer, GND0, connecting to the other computer, SAT0, using Axspawn.

The issue is when SAT0 tries to send something "big" to GND0 only part of the message is being sent.

Here is an example. When GND0 does "ls" in some folder, SAT0 does answer properly.
Screenshot from GND0:
[image]
[image]

Screenshot from SAT0:
[image]

However, when GND0  does "ls -l", only part of what should be sent is being sent.
Screenshot from GND0:
[image]
[image]

Screenshot from SAT0:
[image]

This is what I expect when doing "ls -l":
[image]

Thank you, keep up the great work!!

dranch, 2017-03-19:
This could be an axspawn issue but I see no issue with Direwolf here.  I can propose some ideas for further troubleshooting in the Direwolf support email list but please close this issue as it's out of scope for Direwolf.

felixdube, 2017-03-19:
What is the support email list?

Thank you!!

closed 2017-03-19 by felixdube

dranch, 2017-03-19:
direwolf_packet@yahoogroups.com

dranch, 2017-03-19:
More than you asked but:

The direwolf email support list is is focused on Direwolf and AX.25 issues and can be found at direwolf_packet@yahoogroups.com.  For a serious brain trust of Linux AX.25 users with advanced topics like yours, consider joining the 44net@hamradio.ucsd.edu list.  Though your issue isn't AMPR 44.x.x.x related, I'm SURE someone there will have a thought on your strange issue.  Also consider the Raspberry_Pi_4-Ham_RADIO@yahoogroups.com (if running on an RPI) list as it's quite active.  There is also the  linuxham@groups.io which is a list more focused on Fldigi and it's suite of programs but they support and encourage other Linux + Hamradio application support questions there.  Finally, for more Linux AX.25 kernel centric issues, consider linux-hams@vger.kernel.org (but it's very low traffic these days).  It's really more for reporting and posting fixes for ax.25 kernel issues.

felixdube, 2017-03-19:
Thank you so much for your fast response!!