wb2osz/direwolf #94: Client Side Filter Crashing on WinXP

ncollins805, 2017-03-25:
Using the client side filter to filter IG to RF packets crashes direwolf.  It operates correctly for an indeterminate amount of time before crashing.  Direwolf will still run, but will not transmit to RF and displays windows error.  Using direwolf 1.4 and WinXP SP3.  

[image]
[image]

dranch, 2017-03-25:
What specific version of the 1.4 Beta are you running?  There have been many updates in 2017 which might resolve your issue here.  See what's changed: https://github.com/wb2osz/direwolf/commits/dev .  Unfortunately, there hasn't been any Windows binaries published since 1.4-D found at https://github.com/wb2osz/direwolf/releases/download/1.4-dev-D/direwolf-1.4-dev-D-win.zip .  Maybe you can compile Direwolf on Windows and give the newest version a try?

Can you post the filters you're using so we can see if this can be reproduced on non-Windows machines?  Is there any chance you can test this on a more up to date version of Windows?

--David
KI6ZHD

ncollins805, 2017-03-25:
> 4-dev-D/direwolf-1.4-dev-D-win.zip . Maybe you can compile Direwolf on

The version I'm running is 1.4 dev, just downloaded the other day.  I'm not
at the machine right now so that's the most specific I can be this moment
but the filter is:
FILTER IG 0 !B/K6TJS/N6ACK-11

I can try to compile the newest code when I access the computer.

Nick

ncollins805, 2017-03-26:
I am running version 1.4 Dev-D.
I am unable to compile the dev stream on windows.  I have no experience with this.  I installed make and am getting an error about direwolf.o missing....

dranch, 2017-03-26:
1.4D is the most recent version available for Windows and I'm not familiar with setting up a build environment on Windows.  We might wait to see if John WB2OSZ would be willing to post a new beta build.  If not, do you know for a fact that your copy of DIrewolf 1.4-D will run w/o issue (and duration) without the filter present?

ncollins805, 2017-03-26:
Yes, with the client side filter commented out (disabled), direwolf will
run continuously without error.    Within 2 hours of starting the program
with the client filter enabled, direwolf will crash and must be restarted.
If I comment out the client filter again, direwolf will run without error.
The only thing I changed is the client filter and it started crashing.
 100% correlation.

Nick

wb2osz, 2017-04-02:
I wasn’t able to reproduce this problem on Windows.

Can you try the same thing with the 1.4 beta test version?

 

https://github.com/wb2osz/direwolf/releases

ncollins805, 2017-04-03:
I downloaded and am running the April 2 2017 1.4 Beta.  The IG to transmit
function is now broken.   The IG>TX dialog shows up, but the 0L dialog does
not and will not PTT the radio.  The filter function does not even get
called as the only thing showing up when a packet is received over the
internet is ig>tx.

kd0wlb, 2017-05-09:
Has this been resolved?  i too am having the same thing but i am running win8.  i have noticed that when it does tank it seems to be on an invalid position.  if i disable the position and only use t/osm instead of t/posm in the IGFILTER it runs flawless. Any help would be appreciated.

73

wb2osz, 2017-05-10:
Please include an exact copy of the "invalid position" that seems to be causing trouble.
What is the exact version of the software?
Thanks.

kd0wlb, 2017-05-10:
not sure if it is a position, but in this example any station using all 0's crashes it, there was another i am trying to get it to hang on but they have not beaconed for a bit.

the version i am running is 1.4 dev on a win 8.1 machine
[image]

dranch, 2017-05-10:
I have another local user who I think is also hitting this issue.  When the bad packet comes in, Direwolf doesn't crash on a Raspberry Pi but it stops processing any new incoming packets.  We are trying to see if we can capture the audio that contains the bad packet but it might take some time.

kd0wlb, 2017-05-10:
i have done a little more testing and it looks like it is the telemetry part of the beacon that is causing a problem, i will keep testing and see if i can get it to reproduce with other stations.

dranch, 2017-05-11:
Hello kd0wlb,
Do you see these bad packets often?  If so, could you possibly record some of them to a WAV or MP3 file so we can use it for some testing?

kd0wlb, 2017-05-11:
Hi David,

I  think my issue is the reverse of yours, as i am testing i see that it is traffic coming from the IG to be gated to RF.  I narrowed my IG filter to only a few stations that were close to eliminate others and confine my testing.  when they sent telemetry data via tcp/APRSIS32, Direwolf would lock up and not send any further nor decode but would still function as a part of APRSIS32. When they turned off telemetry i had no failures at all and it kept chugging along even to this morning with no failure.

dranch, 2017-05-11:
Hello JC,
Interesting.  I'm curious, when you disabled telemetry, did you also remove your narrow IG filter to confirm that things were still running ok?

kd0wlb, 2017-05-11:
no i had left the narrow IG filter on to confirm. as soon as i opened it up and had the station transmit telemetry so it could go IG to RF, it crashed Direwolf again.  I am right now running it narrow with no telemetry being transmitted, just position, object and messages.  i tried to use the server side t/pos and leave telemetry off, but it still transmits telemetry even though i am not asking for it and something in Direwolf pops and won't parse just the position and down it goes.

kd0wlb, 2017-07-01:
> Please provide samples of the packets that appear to be causing an issue.
>
> It would also be helpful to run with -dii and -dfff command line options
> to get more details about what is going on in the IGate function and packet
> filtering.

It seems to only happen on my Windblows box.  the pi seems to have no
trouble with them.  here is a screen cap with the switches turned on.  i
opened up the filter to 500km just to get it to error faster while i am
watching it.

[image: Inline image 1]