wb2osz/direwolf #96: KISS port accepts only one connection

regulatre, 2017-04-01:
From what I can tell, I can connect to the KISS port (KISSPORT 8001) but after I disconnect from the KISS port and try to reconnect, connection is refused. It seems Direwolf only accepts a single inbound connection and when that connection closes, no further connections may be made to Direwolf (without restarting direwolf).

na7q, 2017-04-05:
If you're in windows then this is the expected behavior. Linux supports multiple connections at a time. Windows does not. If you are using Linux and having this problem, I'm curious about, as I use multiple KISS connections at any given time, no issues.

nytowl, 2017-05-08:
I wrote a patch that will allow multiple connections , just waiting on the merge request now.

wb2osz, 2017-05-10:
Please try the version on the "dev" branch (1.5 dev A) and see if it is working ok.

closed 2017-06-12 by wb2osz