wb2osz/direwolf #97: Audio Bleed Loop

na7q, 2017-04-08:
I have not had this issue until recently. Could be a problem with the way things are wired, but for quick and simple setups, I would think there'd be a fix for it in the software. I and others, have our transmit audio from Direwolf feeding back into the mic and Direwolf decodes the packet as if it were coming in directly. Usually with an audio level of "0".
Seems Direwolf is hugely sensitive and seems to pick up the bleedthrough even on the lowest audio output settings. I'm using the C119 chipset cards. Applies to every version of Direwolf I've tried, mainly the newest. Would be nice to have a mute during transmit or a filter for sent out packets.

na7q, 2017-04-29:
Hoping for a fix on this soon. It's very annoying for certain types of packet outside of APRS.

na7q, 2017-07-01:
Still looking for some solutions to this.

Luceoria, 2017-10-24:
Assuming you have Linux - check if you have "Auto Gain Control" function set to off in Alsamixer.

na7q, 2017-11-07:
I have it off. Direwolf is just way too sensitive. It's a common problem. But it is what it is.

ab0tj, 2017-11-07:
Just here to add a "me too!" on this one. This had not been a problem until updating to a more recent version. Happens even with AGC off and the transmit audio level way down. Direwolf decodes itself at a low indicated audio level. Maybe this could be solved by having an option to set the modem to half duplex?

isguarded, 2021-05-25:
I have just issued the same problem #337. At a first glance a solution was to use two different hw devices for in- and output.

ADEVICE plughw:0,0

produced the problem. Even when the jacks where connected.

ADEVICE plughw:0,3 plughw:0,0

solved it, given that the connected device is working as it should (quiet). Where 0,3 and 0,0 are ofc the names of your soundcard devices (on linux there should mostly be a seperated device numer for the mic/line-in).

But as you mentioned, the slightest loopback results in a decoding. I use my client app also on fldigi and there is no problem with loopbacks. I don't know what fldigi is doing differently in this regard.

marrold, 2022-12-29:
Looking at the issues, these all seem related:

#401
#97
#288
#337

wb2osz, 2023-05-08:
Fixed - same as 401.

closed 2023-05-08 by wb2osz