wb2osz/direwolf #99: Compiler Warnings under GCC V7.1 and Fedora 26

DwaineGarden, 2017-05-06:
Just posting some of the GCC v7.1 warnings from a compile.

make -j4
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o direwolf.o direwolf.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o config.o config.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o recv.o recv.c
echo " " > tune.h
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD -DGEN_FFF -o gen_fff demod_afsk.c dsp.c textcolor.c -lm -lpthread -lrt -lasound -lgps
echo " " > tune.h
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o dsp.o dsp.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o demod_psk.o demod_psk.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o demod_9600.o demod_9600.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o hdlc_rec.o hdlc_rec.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o hdlc_rec2.o hdlc_rec2.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o multi_modem.o multi_modem.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o rdq.o rdq.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o rrbb.o rrbb.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o dlq.o dlq.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o fcs_calc.o fcs_calc.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o ax25_pad.o ax25_pad.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o ax25_pad2.o ax25_pad2.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o xid.o xid.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o decode_aprs.o decode_aprs.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o symbols.o symbols.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o server.o server.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o kiss.o kiss.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o kissserial.o kissserial.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o kissnet.o kissnet.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o kiss_frame.o kiss_frame.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o hdlc_send.o hdlc_send.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o gen_tone.o gen_tone.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o audio.o audio.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o audio_stats.o audio_stats.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o digipeater.o digipeater.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o cdigipeater.o cdigipeater.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o pfilter.o pfilter.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o dedupe.o dedupe.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o tq.o tq.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o xmit.o xmit.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o morse.o morse.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o ptt.o ptt.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o beacon.o beacon.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o encode_aprs.o encode_aprs.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o latlong.o latlong.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o textcolor.o textcolor.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o dtmf.o dtmf.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o aprs_tt.o aprs_tt.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o tt_user.o tt_user.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o tt_text.o tt_text.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o igate.o igate.c
tt_text.c: In function ‘tt_text_to_call10’:
tt_text.c:542:35: warning: ‘%04d’ directive output may be truncated writing between 4 and 10 bytes into a region of size 8 [-Wformat-truncation=]
  snprintf (stemp, sizeof(stemp), "%04d", packed);
                                   ^~~~
tt_text.c:542:34: note: directive argument in the range [0, 2147483647]
  snprintf (stemp, sizeof(stemp), "%04d", packed);
                                  ^~~~~~
tt_text.c:542:2: note: ‘snprintf’ output between 5 and 11 bytes into a destination of size 8
  snprintf (stemp, sizeof(stemp), "%04d", packed);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o waypoint.o waypoint.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o serial_port.o serial_port.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o log.o log.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o telemetry.o telemetry.c
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD   -c -o dwgps.o dwgps.c
[... cut ...]
This includes support for gpsd.
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD -o ll2utm ll2utm.c geotranz.a textcolor.o misc.a -lm -lpthread -lrt -lasound -lgps
gcc -O3 -pthread -Igeotranz -D_XOPEN_SOURCE=600 -D_DEFAULT_SOURCE=1 -Wall -D_BSD_SOURCE -ffast-math -DUSE_ALSA -DENABLE_GPSD -o utm2ll utm2ll.c geotranz.a textcolor.o misc.a -lm -lpthread -lrt -lasound -lgps
 
Next step - install with:
 
        sudo make install

dranch, 2017-05-06:
Hello Dwaine,
Can you tell us what Linux distribution this is on?

DwaineGarden, 2017-05-06:
Red Hat Fedora 26

DwaineGarden, 2017-05-06:
GCC 7.1.1

DwaineGarden, 2017-05-16:
Did a push for two fixes to the warnings...

wb2osz, 2017-06-12:
Fixed.
Thanks for pointing this out.

closed 2017-06-12 by wb2osz