wb2osz/direwolf #102: PACLEN enforced on V2.0 receipt

rlevine, 2017-06-05:
I think I've found a possible issue with the way direwolf uses PACLEN for v2.0 frames. 

Currently, PACLEN is enforced for the upper bound on the size of both transmitted and received information fields.

In our field usage, PACLEN is set to 128 for field stations, but longer for server BBS TNCs.

To match the TNC behavior we're expecting, it would be helpful for PACLEN to be enforced only for transmitted frames, and not on reciept from the BBS. 

The issue is caused here - ax25_link.c:2212

  	    if (info_len >= 0 && info_len <= S->n1_paclen) 
  	    
Swapping AX25_MAX_INFO_LEN for S->n1_paclen is what I'm after, but a real fix might be to split out params for received and transmitted max lengths in set_version_2_0 () (line 2682) instead of just having one.

Thanks!

Rick

KK6WHJ

wb2osz, 2017-06-05:
Fixed in 'dev' branch.
979fdf1767182b914dd3e306acb4ee7f59d3f1b9

wb2osz, 2017-06-05:
Thanks for pointing this out.

closed 2017-06-05 by wb2osz

rlevine, 2017-06-06:
Thanks for the fix, John!