wb2osz/direwolf #103: [Windows 10] Config file line 202: PTT with RIG is only available when hamlib support is enabled.

VK3FNG, 2017-06-25:
I have Hamrig 1.3 installed, and the DW Binary 1.4.

I get the error "Config file line 202: PTT with RIG is only available when hamlib support is enabled.". I assume that DW has not been compiled with support for hamlib?

Config line in question:

`PTT RIG 122 COM3`

Unfortunately I don't really have any way to key the radio without hamlib. Is there anything I can do?

I tried using FreeBSD but I get audio issues, so I'm back to Windows where at least I can decode :)

wb2osz, 2017-07-01:
hamlib is not currently supported on Windows.
Leaving case open as a reminder to investigate sometime.

radioblues, 2020-11-14:
The steps below are what I did to get Dire Wolf to compile with hamlib support for Windows.

Download the Dire Wolf User Guide

https://github.com/wb2osz/direwolf/blob/master/doc/User-Guide.pdf

Download hamlib for windows:

https://sourceforge.net/projects/hamlib/files/hamlib/3.3/

Run the installer as administrator.

In my case, hamlib installed in C:\Program Files (x86)\hamlib-w64-3.3

Add the hamlib bin directory to the PATH environment variable so that direwolf.exe can find it. Do a search for "windows 10 modify path" if you don’t know how to do that. Alternatively, you can copy the dlls to the same directory where direwolf.exe will be after the steps below.

Add to PATH - C:\Program Files (x86)\hamlib-w64-3.3\bin

Go to section 4.6 in the User Guide and follow the directions for installing cygwin.

Search for and add each of the additional packages and select the version to install. They are under the devel category. I selected what looks like the latest stable versions.

Note that in the User Guide, the package bin-utils should be binutils and that the line "cd buile" should be "cd build".

In addition to the export lines in the user guide, the following line updates the Cygwin bash PATH environment variable to provide access to the hamlib files during the build process:

export PATH=$PATH:/cygdrive/c/Program\ Files\ \(x86\)/hamlib-w64-3.3/include/:/cygdrive/c/Program\ Files\ \(x86\)/hamlib-w64-3.3/lib/gcc

After building, testing, and creating a zip file, copy the zip file to the desktop and decompress it.

cp ~/direwolf/build/direwolf*.zip  /cygdrive/c/Users/LUSER/Desktop/direwolf

From there, configure the Dire Wolf to use hamlib and the IC-7100:

PTT RIG 370 COM3

dranch, 2024-02-26:
As requested by WB2OSZ to ensure this is tracked, Joe K0OG kindly uploaded his Direwolf v1.7 for Windows with Hamlib support here:

> I built Direwolf 1.7 (final release) with Hamlib 4.5.5 (latest stable), and posted it here:
>
> https://www.dropbox.com/scl/fo/8e49jwuchr4zccwgwh0db/h?rlkey=oynzlubqef9lj3njwbuiy8c3d&dl=0

On the long term, hopefully the official Direwolf Windows builds will include Hamlib support.