wb2osz/direwolf #104: PTT configuration doesn't support long /dev names

dranch, 2017-07-04:
I'm using a US Interface Navigator here and I'm trying to use the serial by-id naming convention.  The Navigator's six serial ports identify themselves with long names:

root@rpi3:/etc/ax25# ls -la /dev/serial/by-id/
total 0
drwxr-xr-x 2 root root 160 Jul  4 10:42 .
drwxr-xr-x 4 root root  80 Jul  4 10:42 ..
lrwxrwxrwx 1 root root  13 Jul  4 10:42 usb-FTDI_Navigator__CAT___2nd_PTT__00000000-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root  13 Jul  4 10:42 usb-FTDI_Navigator__CAT___2nd_PTT__00000000-if01-port0 -> ../../ttyUSB1
lrwxrwxrwx 1 root root  13 Jul  4 10:42 usb-FTDI_Navigator__RS232___Config__00000002-if00-port0 -> ../../ttyUSB4
lrwxrwxrwx 1 root root  13 Jul  4 10:42 usb-FTDI_Navigator__RS232___Config__00000002-if01-port0 -> ../../ttyUSB5
lrwxrwxrwx 1 root root  13 Jul  4 10:42 usb-FTDI_Navigator__WKey___FSK__00000001-if00-port0 -> ../../ttyUSB2
lrwxrwxrwx 1 root root  13 Jul  4 10:42 usb-FTDI_Navigator__WKey___FSK__00000001-if01-port0 -> ../../ttyUSB3

The port I'm trying to use is:

lrwxrwxrwx 1 root root  13 Jul  4 10:42 usb-FTDI_Navigator__CAT___2nd_PTT__00000000-if01-port0 -> ../../ttyUSB1

and thus in the direwolf.conf file for Direwolf v1.4, I'm using:

   PTT /dev/serial/by-id/usb-FTDI_Navigator__CAT___2nd_PTT__00000000-if01-port0 RTS -DTR

Unfortunately, that doesn't work but the short direct interface name DOES work:

   PTT /dev/ttyUSB1 RTS -DTR

Can Direwolf be enhanced to deal with longer device names?

--David
KI6ZHD

tnalpgge, 2017-10-07:
I tried this today with a very long device name and ran into it.

```
: [username@workstation ~]$ ; grep PTT direwolf.conf
#PTT /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_D-if00-port0 RTS DTR
PTT /dev/serial/by-path/pci-0000:00:14.0-usb-0:4:1.0-port0 RTS DTR

: [username@workstation ~]$ ; direwolf
Dire Wolf version 1.2
Audio device for both receive and transmit: sysdefault:CARD=PCH  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate.
ERROR can't open device /dev/serial/by-path/ for channel 0 PTT control.
No such file or directory
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Ready to accept AGW client application 0 on port 8000 ...
Use -p command line option to enable KISS pseudo terminal.
Ready to accept KISS client application on port 8001 ...
```

Yes, I tried both device paths showed by the `grep` but both end up with the funny unicode character in the output and no working PTT.

This was encountered on:
```
: [username@workstation ~]$ ; uname -a
Linux workstation 4.4.0-96-generic #119-Ubuntu SMP Tue Sep 12 14:59:54 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
: [username@workstation ~]$ ; cat /etc/os-release 
NAME="Ubuntu"
VERSION="16.04.3 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.3 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
```

About the `.deb` from which I installed the software:

```
: [username@workstation ~]$ ; apt show direwolf
Package: direwolf
Version: 1.2-2build1
Priority: optional
Section: universe/hamradio
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Hamradio Maintainers <debian-hams@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 912 kB
Depends: libasound2 (>= 1.0.16), libc6 (>= 2.15), libgps22 (>= 3.3), direwolf-docs
Suggests: gpsd
Breaks: direwolf-docs (<< 1.1-1)
Replaces: direwolf-docs (<< 1.1-1)
Homepage: http://home.comcast.net/~wb2osz/site/
Download-Size: 244 kB
APT-Manual-Installed: yes
APT-Sources: http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 Packages
Description: Soundcard TNC for APRS
 Dire Wolf is a software "soundcard" modem/TNC and APRS encoder/decoder. It can
 be used stand-alone to receive APRS messages, as a digipeater, APRStt gateway,
 or Internet Gateway (IGate). It can also be used as a virtual TNC for other
 applications such as APRSIS32, UI-View32, Xastir, APRS-TW, YAAC, UISS, Linux
 AX25, SARTrack, and many others.
```

dranch, 2017-10-08:
While I don't think this issue has been resolved in Direwolf 1.5 (GIT tip of tree) but the Direwolf version you're running is VERY old.  It's highly recommended you use Direwolf v1.4.

--David

wb2osz, 2017-10-08:
Fixed in commit 33a34f3466dda77cb5de25f0db2441ccee710bd2 on the "dev" branch which is leading up to release 1.5.

To get the most recent version:

git clone http://github.com/wb2osz/direwolf
cd direwolf
git checkout dev
make
sudo make install

closed 2017-10-08 by wb2osz