wb2osz/direwolf #105: Digipeat "Erase

na7q, 2017-07-14:


na7q, 2017-07-17:
The REGEN replacement of DIGIPEAT does what I need, however it does not follow any filters, or anything in the config file. So still a bust.

craigerl, 2019-03-01:
you can look into REGEN, but with no "REGENFILTER" implemented it would get out of control really fast.  I agree with your feature Idea.  I have a portable that is completely deaf to the world and the only thing it can hear is my digipeater.  It's ultimately useless when the internet goes out.    The digipeater hears the message replies that are supposed to go to my portal, and does nothing to get them there.

craigerl, 2019-03-02:
maybe a REGEN that puts NORF or NOGATE or something in the header to make sure there is never an infinite feedback loop?  i cloned the source, but the solution wasnt obvious.   regenerating packets for a specific callsign inside the packet data would be a killer feature for me in the mountains where ht radios are otherwise useless.  a regen would allow my ht to leverage the big digipeater antenna.

na7q, 2021-06-23:
Still seems there is a small bit of demand for this feature. I'm wondering if the filtering functions will be supported with this at some point? Such as DEDUPE.

wb2osz, 2023-04-10:
See User Guide, "Frame Regeneration" section.
Use this cautiously only for special situations.  Retransmitting on the same frequency would be a bad idea.  If two people did this, a packet could bounce back and forth between them forever.
Closing.

closed 2023-04-10 by wb2osz