wb2osz/direwolf #111: Dire wolf 1.4 No Digipeating

KD8YNY, 2017-08-19:
I installed 1.4 last night and set it up today, it would seem that it is not digipeating at all.  I see information flooding my screen "[ig>tx] KE8NK-3>APN391,qAR,KD8KTS-10:!3916.60NS08100.02W#PHG3560 W2,WVn-N, Ritchie Co, WV" but the information is not being sent out over the radio.  The radio will transmit when Dire Wolf sends out its location info, it receives local transmissions but nothing gets repeated out...

I have gone over the config file multiple times and can't see any issues..

dranch, 2017-08-19:
Github issues aren't for getting help with Direwolf.  Please join the Direwolf Yahoo group at direwolf_packet@yahoogroups.com.  Anyway, to conclude this issue, your issue isn't a digipeating issue in the technical sense as that should be a packet heard on RF and resent on RF.  Your specific data packet is coming from the Internet (Igate) and I assume you're issue is that it's not going out on RF?  This Igate to RF repeat generally does NOT happen for any APRS data packets except for APRS messages going to remote stations that have been recently heard by *your* Direwolf setup, broadcast type APRS bulletins, etc.  General APRS posits like this should generally be not repeated on RF or your station would be transmitting 100% every day!   Have you read https://github.com/wb2osz/direwolf/blob/master/doc/Successful-APRS-IGate-Operation.pdf , it talks about the "IGTXVIA" parameter.  Please join the above Yahoo group and in your first message, tell the group what have you configured for this parameter.

--David
KI6ZHD

KD8YNY, 2017-08-19:
Thanks for the reply, it was more of a wondering if this was a big sort of post.  My last direwolf 1.2 broadcast from the internet over my machine within a set range of miles...
Anyway, i will get on that group and talk more on there about it..

nickoe, 2018-02-09:
... I guess this issue can be closed.

wb2osz, 2018-02-10:
The default filter for IS to RF IGate was changed.  The default is now appropriate for general use.

closed 2018-02-10 by wb2osz