wb2osz/direwolf #112: Support APRS Directed Station Queries

dranch, 2017-09-04:
Direwolf 1.4 recognizes APRS station queries but doesn't respond as the "aprs_directed_station_query" function is effectively just a placeholder and no code support exists today.  I would be great to add support for the basic commands.  Here is a proposal of maybe how they could be supported over time.  Just my $0.02 here..

Phase 1 - have Direwolf just respond with "complete response not supported today".  Just getting a response will be a big help for users trying to do diagnostics, etc.

Future proposed phases:

Query Type | Query
APRS | General — All stations query | Station’s position and status | phase 2: should be strait forward
--
APRSD | Directed — Query an individual station for stations heard direct | List of stations heard direct | Could be a Phase3 feature though this is a pretty valuable response
--
APRSH | Directed — Query if an individual station has heard a particular station | Position of heard station as an APRS Object, plus heard statistics for the last 8 hours | Maybe a phase 4 feature
--
APRSM | Directed — Query an individual station for outstanding unacknowledged or undelivered
messages | All outstanding messages for the querying station | Maybe a phase 4 feature
--
APRSO | Directed — Query an individual station for its Objects | Station’s Objects | maybe phase 3
--
APRSP | Directed — Query an individual station for its position | Station’s position | phase 2: should be strait forward
--
APRSS  | Directed — Query an individual station for its status | Station's status | phase 2: should be strait forward
--
APRST or
PING? | Directed — Query an individual station for a trace (i.e. path by which the packet was heard) | Route trace | Could be a Phase4 feature though this is a pretty valuable response
--
IGATE | General — Query all Internet Gateways | IGate station capabilities | phase 2: should be strait forward
--
WX | General — Query all weather stations | Weather report (and the station’s position if it is not included in the Weather Report) | Maybe phase 3 Seems to be a subset of APRSO

dranch, 2017-09-04:
It's worth adding that Direwolf currently digipeats these packets which is probably NOT what we want to have happen.  This should be suppressed as part of the phase 1effort

Tyler-2, 2022-06-09:
Yes please! I didn't even know about these features since Direwolf is kind of my golden standard for APRS, and where I learn.

craigsned, 2023-01-26:
Bump - would be a really useful addition to Direwolf