wb2osz/direwolf #115: Per Channel iGating

nigelvh, 2017-10-04:
I have a setup where I'd like to use a different ID and potentially a different server when iGating packets from a different channel. I haven't been able to find a way to accomplish this in the users guide as it stands. Is this a possibility?

na7q, 2017-11-07:
You could run to instances of Direwolf. Two configurations and this should work. However I have distortion on transmit when running two sound cards. Maybe you'll have better luck with that. Otherwise you can easily do the 2 instances of it with 2 configuration files.

nigelvh, 2017-11-07:
Is it possible to run two instances using the two channels on one sound card? The current setup uses the left channel for one radio, and the right for the other. It keeps the cabling fairly tidy, and removes the question as to what USB device gets enumerated first.

I'm kind of guessing not, which is why I opened the issue ticket. It's entirely possible that it's just not something that has wide enough utility to be valuable enough to add to the program, but I figured it wouldn't hurt to put the idea out there.

dranch, 2017-11-07:
Github is used for reporting bugs or requesting enhancements.  For general support, please subscribe and post to direwolf_packet@yahoogroups.com .  As to your specific question, no, Direwolf takes control of the entire sound card but it DOES allow for creating unique CHANNELS for the left and right channels on a sound card.  The bigger issue is if your soundcard doesn't offer stereo microphone inputs, you'll only be able to transmit for one CHANNEL and the other will have to be RX only.  If you have more questions, please ask them on the Yahoo list and close this ticket.

nigelvh, 2017-11-07:
David, I think there's been a bit of a miscommunication here. I have a single sound card, using the left and right channels for two different receivers, and I iGate from both of them to the internet.

I would like to be able to iGate each of them with a different callsign (specifically just want to be able to change the -X ID number on the callsign, but generalizing it to the whole callsign makes sense to me).

Thank you for confirming my theory that I could not run two instances of direwolf on the same sound card, which leaves intact my original enhancement request to allow for per channel iGate configs.