wb2osz/direwolf #117: Can Direwolf provide a low level external interface into layer2?

kantooon, 2017-10-09:
Hi,
I have a question: I want to use an external application to handle layer1 (radio interface) and connect it with Direwolf at an arbitrary bitrate (125 kbit/s to 2 Mbit/s). I would like Direwolf to handle layer2 and above (AX25 etc.)
Can Direwolf provide me with an interface into layer2, skipping everything related to modulation and demodulation? It can be anything, a UNIX socket, TCP/IP socket (preferred for remote radio interface) pts device etc.?
I would like to use this magnificent piece of software instead of reinventing the wheel.
Please see the architecture of this stack for something similar of what I have in mind: https://osmocom.org/projects/osmobts/wiki
Thanks,
Adrian

wb2osz, 2017-10-10:
It's currently not possible to bypass the internal "soundcard" modem and use an external modem.  This topic has come up a couple times before.

Would a "KISS" interface (both serial port & TCP) be suitable?  i.e. The external modem would look like a KISS TNC and transfer units of frames?

Please contact me directly.  wb2osz-at-comcast-dot-net

dranch, 2017-10-10:
Github is not a support mechanism, please join direwolf_packet@yahoogroups.com for general support.  Please close this ticket.  As so your question (it's been asked before btw), no, direwolf IS the modem and the protocol stack.  It does support KISS to say a Linux kernel AX.25 stack though.. might work for you.

--David
KI6ZHD

closed 2017-10-10 by kantooon