wb2osz/direwolf #121: Problems with KISS over TCP

wbustraan, 2017-11-17:
I am working with the latest version (commit 590c82b) of Direwolf on a Mac. There seems to be some problems with transmitting packets via the KISS TCP interface. I am working on a client application that connects to the KISS port and sends packets. After struggling to figure out what was wrong with my code, I began to suspect that maybe the problem is with Direwolf. So I took my app out of the equation by using kissutil to format and send the packets instead and discovered an issue.

To reproduce the issue, I created a text file with a sample packet found in the User Guide:
`K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630`

I started Direwolf:
`direwolf -t 0 -d kn`
and then also started kissutil:
`./kissutil -f ~/tmp/kisstest`
I then copied the sample packet into the transmit directory, sequentially, 4 times, waiting each time for the audio to let me know that the packet had been processed. In addition, I ran Wireshark in order to capture the packets passing over the loopback interface, just to ensure that the bytes produced by kissutil were exactly the same.

Even though it is the exact same packet of data, you can see from the log that Direwolf does not process it the same way all 4 times. 

Here is the output from kissutil:
> Processing kisstest2.txt for transmit...
> K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630
> [0] K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630
> Processing kisstest2.txt for transmit...
> K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630
> [0] K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630
> Processing kisstest2.txt for transmit...
> K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630
> Processing kisstest2.txt for transmit...
> K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630
> [0] K1NRO-1>APD14p-7,WIDE2-2:!4238.80NS07105.63W#PHG5630

Here is the output from direwolf:
```Dire Wolf DEVELOPMENT version 1.5 B (Nov 16 2017)

Reading config file /Users/wbustraa/direwolf.conf
Audio input device for receive: Built-in Microphone:0  (channel 0)
Audio out device for transmit: Built-in Output:1  (channel 0)
Number of devices = 5
--------------------------------------- device #0
[ Default Input ]
Name        = "Built-in Microphone"
Host API    = Core Audio
Max inputs  = 2
Max outputs = 0
--------------------------------------- device #1
[ Default Output ]
Name        = "Built-in Output"
Host API    = Core Audio
Max inputs  = 0
Max outputs = 2
--------------------------------------- device #2
Name        = "Soundflower (2ch)"
Host API    = Core Audio
Max inputs  = 2
Max outputs = 2
--------------------------------------- device #3
Name        = "Soundflower (64ch)"
Host API    = Core Audio
Max inputs  = 64
Max outputs = 64
--------------------------------------- device #4
Name        = "Multiple Output"
Host API    = Core Audio
Max inputs  = 0
Max outputs = 2
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate.
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Use -p command line option to enable KISS pseudo terminal.
Ready to accept AGW client application 0 on port 8000 ...
Ready to accept KISS TCP client application 0 on port 8001 ...

Attached to KISS TCP client application 0...

Ready to accept KISS TCP client application 1 on port 8001 ...

<<< Data frame from KISS client application, port 0, total length = 53
  000:  c0 00 82 a0 88 ae 62 68 e0 96 62 9c a4 9e 40 e2  ......bh..b...@.
  010:  ae 92 88 8a 64 40 65 03 f0 21 34 32 33 38 2e 38  ....d@e..!4238.8
  020:  30 4e 53 30 37 31 30 35 2e 36 33 57 23 50 48 47  0NS07105.63W#PHG
  030:  35 36 33 30 c0                                   5630.
[0L] K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630

K1NRO-1 audio level = 6(3/2)   [NONE]   ||||||___
[0.2] K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630
Position, OVERLAY DIGI (green star) w/overlay S, DireWolf, WB2OSZ, 25 W height=640 3dBi omni
N 42 38.8000, W 071 05.6300

>>> Data frame to KISS client application, port 0, total length = 53
  000:  c0 00 82 a0 88 ae 62 68 e0 96 62 9c a4 9e 40 e2  ......bh..b...@.
  010:  ae 92 88 8a 64 40 65 03 f0 21 34 32 33 38 2e 38  ....d@e..!4238.8
  020:  30 4e 53 30 37 31 30 35 2e 36 33 57 23 50 48 47  0NS07105.63W#PHG
  030:  35 36 33 30 c0                                   5630.

<<< Data frame from KISS client application, port 0, total length = 53
  000:  c0 00 82 a0 88 ae 62 68 e0 96 62 9c a4 9e 40 e2  ......bh..b...@.
  010:  ae 92 88 8a 64 40 65 03 f0 21 34 32 33 38 2e 38  ....d@e..!4238.8
  020:  30 4e 53 30 37 31 30 35 2e 36 33 57 23 50 48 47  0NS07105.63W#PHG
  030:  35 36 33 30 c0                                   5630.
[0L] K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630

K1NRO-1 audio level = 5(2/2)   [NONE]   ||||||||_
[0.3] K1NRO-1>APDW14,WIDE2-2:!4238.80NS07105.63W#PHG5630
Position, OVERLAY DIGI (green star) w/overlay S, DireWolf, WB2OSZ, 25 W height=640 3dBi omni
N 42 38.8000, W 071 05.6300

>>> Data frame to KISS client application, port 0, total length = 53
  000:  c0 00 82 a0 88 ae 62 68 e0 96 62 9c a4 9e 40 e2  ......bh..b...@.
  010:  ae 92 88 8a 64 40 65 03 f0 21 34 32 33 38 2e 38  ....d@e..!4238.8
  020:  30 4e 53 30 37 31 30 35 2e 36 33 57 23 50 48 47  0NS07105.63W#PHG
  030:  35 36 33 30 c0                                   5630.

<<< Rejected Noise from KISS client application, total length = 3
  000:  00 82 a0                                         ...

<<< TXtail from KISS client application, port 6, total length = 34
  000:  c0 64 40 65 03 f0 21 34 32 33 38 2e 38 30 4e 53  .d@e..!4238.80NS
  010:  30 37 31 30 35 2e 36 33 57 23 50 48 47 35 36 33  07105.63W#PHG563
  020:  30 c0                                            0.
KISS protocol set TXtail = 64 (*10mS units = 640 mS), port 6

<<< Data frame from KISS client application, port 0, total length = 53
  000:  c0 00 82 a0 88 62 68 e0 ae 96 62 9c a4 9e 40 e2  .....bh...b...@.
  010:  ae 92 88 8a 64 40 65 03 f0 21 34 32 33 38 2e 38  ....d@e..!4238.8
  020:  30 4e 53 30 37 31 30 35 2e 36 33 57 23 50 48 47  0NS07105.63W#PHG
  030:  35 36 33 30 c0                                   5630.
[0L] K1NRO-1>APD14p-7,WIDE2-2:!4238.80NS07105.63W#PHG5630
Destination Address has lower case letters. "APD14p-7" must be all upper case.

*** The origin and journey of this packet should receive some scrutiny. ***

K1NRO-1 audio level = 6(3/2)   [NONE]   _||||||__
[0.3] K1NRO-1>APD14p-7,WIDE2-2:!4238.80NS07105.63W#PHG5630
Position, OVERLAY DIGI (green star) w/overlay S, 25 W height=640 3dBi omni
N 42 38.8000, W 071 05.6300
Destination Address has lower case letters. "APD14p-7" must be all upper case.

*** The origin and journey of this packet should receive some scrutiny. ***

>>> Data frame to KISS client application, port 0, total length = 53
  000:  c0 00 82 a0 88 62 68 e0 ae 96 62 9c a4 9e 40 e2  .....bh...b...@.
  010:  ae 92 88 8a 64 40 65 03 f0 21 34 32 33 38 2e 38  ....d@e..!4238.8
  020:  30 4e 53 30 37 31 30 35 2e 36 33 57 23 50 48 47  0NS07105.63W#PHG
  030:  35 36 33 30 c0                                   5630.```

wb2osz, 2018-02-12:
I don't have a Mac and have not been able to reproduce this problem on Linux.
However, a bug that caused some strange behavior with KISS over TCP was fixed recently.
Can you try the most recent version on the "dev" branch and see if the issue persists?
Thanks.

jcrix, 2018-03-26:
I also found the same problem. You can clearly watch direwolf not pass the messages over the KISS TNC connection.  Its sometimes almost every other packet.  I've used 1.5 and the current beta test 2.  This does not effect AGW protocol. It only does it on KISS. I'm running linux.

jcrix, 2018-03-27:
As it turns out. My problem was related to incorrectly handling parts of the protocol and timing that caused my issues.  Its working fine now after realizing that.  I'm guessing there's logic in direwolf to kinda ignore things when they aren't expected. Cheers!

wb2osz, 2023-04-10:
Is there any reason to keep this open any longer?

dandidev, 2024-09-17:
Hello! I use versions 1.7 and 1.8 of direwolf. It sends data to kissserial, this data is logged by direwolf. However, it does not send it on to kiss TCP.
I think that data sent to kissserial should be forwarded to direwolf over tcp. Am I getting this right?
Is it possible that I made a mistake and it depends on the setting?
OS: archlinux,
direwolf version: 1.7, 1.7.1, 1.8
If I understand correctly, this is a similar bug to this bug ticket. But correct me if I'm wrong!

Thx

dranch, 2024-09-17:
@dandidev - Your issue seems to be different than this Github issue ticket and github issues are generally only for reporting bugs or enhancement requests.  Please repost your issue to the direwolf@groups.io list to get help.  In that email, please detail your scenario on what kind of packets you're seeing/not-seeing be it connected or non-connected packets, etc.

wb2osz, 2025-03-30:
2 years ago, I asked, "Is there any reason to keep this open any longer?" and no reply.
Closing issue.

closed 2025-03-30 by wb2osz