wb2osz/direwolf #122: EASYDIGI PTT issue

lmcycles, 2017-11-22:
I've tried to search for a resolution, but all my searches result in other with same issue.

I have a Easy Digit built and connected to my Yeasu FT-212RH and in direwolf when I un-comment the PTT /dev/ttyUSB0 and start Direwolf the radio immediately keys up and starts transmitting.

What am I missing here?

Regards,
KE0LFX

dranch, 2017-11-22:
Github's "issue" area is not for support.. it's for reporting bugs or feature enhancements.  I recommend to join direwolf_packet@yahoogroups.com to get support.  Until then, I would suspect either you're using the wrong pin on your RS232 serial adapter or you have the wrong logic configured in your direwolf.conf.

--David
KI6ZHD

dranch, 2017-11-22:
Ps.  Please close this issue once you have your issued resolved via help from the Yahoo group.  :-)

lmcycles, 2017-11-22:
My apologies David,

I will move the message over to the Yahoo Group....

Thanks.

Regards,
Jeremy

closed 2017-11-22 by lmcycles