wb2osz/direwolf #127: Including Digipeater Path In Logging Output

wsmitchell3, 2018-01-16:
Recently I've been working on some projects where the digipeater path that a packet takes is of interest.  While I would love to just use the logging feature to store the information I need, it doesn't seem to include that.  Could the digipeater path be added to the logging output?

For the time being I have been recording the audio, then use atest.exe to put the raw output into a file.  I suppose I could also redirect the output stream, but that has other disadvantages (including not being able to see it live).

Here are a few examples of packets I'm looking at (from atest.exe output), and they include some IGTX packets (callsigns and info changed):
```
DECODED[42] Digipeater WIDE2 audio level = 52(25/14)     
[0] K0XAB-1>APMI04,W0XCD-1,WIDE2*:}K0XEF-4>APNU19,TCPIP,K0XAB-1*:!0023.00NS00015.00W#PHG7630/ (K0XEF-4 W3,MNn Digipeaterville,MN UIDIGI 1.9B3) 2

DECODED[66] Digipeater WIDE3 audio level = 52(27/13)     
[0] K0XGH-10>APNU19,N0XIJ-1,K0XAB-1,W0XCD-1,WIDE3*:!0026.00NS00015.20W#PHG7280/(K0XGH-10 W3,MNn Digitown,MN) 2
```

In the log, I would like to see these recorded in some format like this, to enable offline analysis and processing:
```
K0XAB-1>APMI04,W0XCD-1,WIDE2*:}K0XEF-4>APNU19,TCPIP,K0XAB-1*

K0XGH-10>APNU19,N0XIJ-1,K0XAB-1,W0XCD-1,WIDE3*
```

wb2osz, 2018-01-17:
Take a look at the new "kissutil" application in version 1.5 beta test.
Does this help?

wsmitchell3, 2018-01-17:
That's certainly a step in the right direction, thanks!.
Part of what I'd like is the timestamp with the packets.  I guess it would work to enable the direwolf log, pipe the kissutil output to another file, and cross-correlate.

closed 2018-01-17 by wsmitchell3