wb2osz/direwolf #128: raspberry pi and dev version 1.5

ve1jot, 2018-01-17:
Just tried it, and it will not start, maybe I am missing something that has changed in conf file or something?
"direwolf: multi_modem.c:319: multi_modem_process_sample: Assertion `save_audio_config_p->achan[chan].num_subchan > 0 && save_audio_config_p->achan[chan].num_subchan <= 9' failed."

ve1jot, 2018-01-20:
got it to run by remarking out lines 319 and 320 in multi_modem.c, but not sure what effect if any this will have...

wb2osz, 2018-01-23:
Something is seriously wrong if that check fails.
What is in your configuration file?

ve1jot, 2018-01-23:
It is the exact same config file that works fine from compiled master branch 1.4..
#############################################################
#                                                           #
#               Configuration file for Dire Wolf            #
#                                                           #
#                   Linux version                           #
#                                                           #
#############################################################
#
# Consult the User Guide for more details on configuration options.
#
#
# These are the most likely settings you might change:
#
#	(1)   	MYCALL 	-  call sign and SSID for your station.
#
#			Look for lines starting with MYCALL and 
#			change NOCALL to your own.
#
#	(2)	PBEACON	-  enable position beaconing.
#
#			Look for lines starting with PBEACON and 
#			modify for your call, location, etc.
#
#	(3)	DIGIPEATER  -  configure digipeating rules.
#
#			Look for lines starting with DIGIPEATER.
#			Most people will probably use the given example.
#			Just remove the "#" from the start of the line
#			to enable it.
#
#	(4)	IGSERVER, IGLOGIN  - IGate server and login
#
#			Configure an IGate client to relay messages between 
#			radio and internet servers.
#
#
# The default location is "direwolf.conf" in the current working directory.
# On Linux, the user's home directory will also be searched.
# An alternate configuration file location can be specified with the "-c" command line option.  
#
# As you probably guessed by now, # indicates a comment line.
#
# Remove the # at the beginning of a line if you want to use a sample
# configuration that is currently commented out.
#
# Commands are a keyword followed by parameters.
#
# Command key words are case insensitive.  i.e. upper and lower case are equivalent.
#
# Command parameters are generally case sensitive.  i.e. upper and lower case are different.
#

#############################################################
#                                                           #
#               FIRST AUDIO DEVICE PROPERTIES               #
#               (Channel 0 + 1 if in stereo)                #
#                                                           #
#############################################################

#
# Many people will simply use the default sound device.
# Some might want to use an alternative device by chosing it here.
#
# Linux ALSA is complicated.  See User Guide for discussion.
# To use something other than the default, generally use plughw
# and a card number reported by "arecord -l" command.  Example:

ADEVICE  plughw:1,0

[... 368 lines cut ...]
# This is in 10 ms units so 100 means 1000 ms = 1 second.

#TTOBJ 0 1 WIDE1-1
#CHANNEL 0
#DWAIT 100

# Advertise gateway position with beacon.

# OBEACON DELAY=0:15 EVERY=10:00 VIA=WIDE1-1 OBJNAME=WB2OSZ-tt SYMBOL=APRStt LAT=42^37.14N LONG=71^20.83W COMMENT="APRStt Gateway"  

# Sample speech responses.
# Default is Morse code "R" for received OK and "?" for all errors.

#TTERR  OK               SPEECH  Message Received.
#TTERR  D_MSG            SPEECH  D not implemented.
#TTERR  INTERNAL         SPEECH  Internal error.
#TTERR  MACRO_NOMATCH    SPEECH  No definition for digit sequence.
#TTERR  BAD_CHECKSUM     SPEECH  Bad checksum on call.
#TTERR  INVALID_CALL     SPEECH  Invalid callsign.
#TTERR  INVALID_OBJNAME  SPEECH  Invalid object name.
#TTERR  INVALID_SYMBOL   SPEECH  Invalid symbol.
#TTERR  INVALID_LOC      SPEECH  Invalid location.
#TTERR  NO_CALL          SPEECH  No call or object name.
#TTERR  SATSQ            SPEECH  Satellite square must be 4 digits.
#TTERR  SUFFIX_NO_CALL   SPEECH  Send full call before using suffix.

ve1jot, 2018-01-23:
Egad, cut and paste did not have the desired effect...

ve1jot, 2018-01-23:
ok, did a fresh pull, compiled dev 1.5, and now it appears to work(from console):
Dire Wolf version 1.5 (Jan 23 2018) Beta Test
Includes optional support for:  cm108-ptt

Reading config file /home/ve1jot/direwolf.conf
Config file: Beaconing should be configured for channel 0 when digipeating is enabled.
Audio device for both receive and transmit: plughw:1,0  (channel 0)
Channel 0: 300 baud, AFSK 1600 & 1800 Hz, D, 44100 sample rate / 4.
        0.0: D 1585 & 1785
        0.1: D 1615 & 1815
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Use -p command line option to enable KISS pseudo terminal.
Ready to accept KISS TCP client application 0 on port 8001 ...
Ready to accept AGW client application 0 on port 8000 ...
-------------------------------------------------------------------------------------------------

Now I will attempt it after an install so it *should* come right up during a reboot..I'm honestly now not sure why it did error that time..the direwolf.conf file hasn't been changed...if it works now, sorry for noise hi!

ve1jot, 2018-01-23:
Ok, comes right up now...honestly not sure what caused the error in the first place, possibly dirty usb connection on the sound dongle on the pi..as the only thing I did was unplug and re-insert the dongle..

closed 2018-01-23 by ve1jot